pietern/goestools #8: Detect nanomsg drops

pietern, 2018-03-09:
Right now nanomsg drops are silent. If packets are missing from the stream it is not possible to tell whether this is happening on the downlink or on the network. Detection for this should be added.

pietern, 2018-06-01:
This is still a problem in theory but with the much larger buffer size it's not a problem in practice.

Ignoring and closing.

closed 2018-06-01 by pietern

eminence, 2019-06-06:
I think I might be running into this, due to a poor wifi link between goesrecv and goesproc.  My symptom is lots of "TP_PDU drop" in goesproc, but goesrecv is reporting a very constant number of received and a very lot number of dropped packets (so I don't think I'm running into #26).

So I'm interested in working on detecting this type of packet drops.  Do you have any suggestion about an approach to detect nanomsg packet drops?  Does nanomsg have some built-in functionality to handle this?  Or maybe the first few bytes of the packet could be used for a packet number counter?