pietern/goestools #11: DCS processing

pietern, 2018-04-03:
This could be nice. Using this issue to collect some links.

* http://eddn.usgs.gov/dcpformat.html
* https://acwi.gov/hydrology/stiwg/index.html

dot-bob, 2018-09-10:
Here are a couple more useful links:
- [https://noaasis.noaa.gov/DCS/index.html](https://noaasis.noaa.gov/DCS/index.html)
- [http://eddn.usgs.gov](http://eddn.usgs.gov) - A site the USGS uses to manually retrieve DCS data.

I used to be pretty involved in DCS a few years back, primarily on the transmitting side to the satellite.  Most of the DCS data is from gauging stations like rain fall, air temperature, stream flow, river level, lake level, etc...

TWG meetings occasionally have interesting details that are upcoming with the GOES satellites   If you have any questions feel free to ask.

pietern, 2019-02-25:
New DCS format documentation: https://dcs1.noaa.gov/documents/HRIT%20DCS%20File%20Format.pdf

tmbates12, 2019-05-19:
There was a revision to this documentation early this year, mostly clearing up some ambiguities as well as some incorrect information: https://dcs1.noaa.gov/documents/HRIT%20DCS%20File%20Format%20Rev1.pdf

tmbates12, 2019-07-25:
Here's an example of a transmitter used for stream gauge measurements. It seems that the data is sent from a data processing board over serial to the transmitter, allowing for easily replaceable parts.
ftp://ftp.vaisala.com/sunsoft/555_windows_software/manuals/sattrans.pdf

tmbates12, 2019-07-26:
Here's a document about some of the most common data formats I found in the DCP Data.
https://www.sutron.com/wp-content/uploads/2014/01/36e86d82613460cb3b4cf364c4daf2f4.pdf