pietern/goestools #19: new raspian version make command error

hankbillings, 2018-07-09:
For some reason I'm getting the following error after using the make command and following the instructions line by line on raspbian lite and/or raspbian desktop.

src/goesrecv/CMakeFiles/rtlsdr_source.dir/build.make:62: recipe for target 'src/goesrecv/CMakeFiles/rtlsdr_source.dir/rtlsdr_source.cc.o' failed
make[2]: *** [src/goesrecv/CMakeFiles/rtlsdr_source.dir/rtlsdr_source.cc.o] Error 1
CMakeFiles/Makefile2:2246: recipe for target 'src/goesrecv/CMakeFiles/rtlsdr_source.dir/all' failed
make[1]: *** [src/goesrecv/CMakeFiles/rtlsdr_source.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

pietern, 2018-07-09:
Hi, thanks for giving it a try. Can you include error messages just above the ones you included?

hankbillings, 2018-07-10:
Thank you... Below is the full error...

[ 78%] Building CXX object src/goesrecv/CMakeFiles/rtlsdr_source.dir/rtlsdr_source.cc.o
/home/hankbillings/goestools/src/goesrecv/rtlsdr_source.cc: In member function ‘void RTLSDR::setFrequency(uint32_t)’:
/home/hankbillings/goestools/src/goesrecv/rtlsdr_source.cc:75:8: warning: unused variable ‘rv’ [-Wunused-variable]
   auto rv = rtlsdr_set_center_freq(dev_, freq);
        ^~
/home/hankbillings/goestools/src/goesrecv/rtlsdr_source.cc: In member function ‘void RTLSDR::setSampleRate(uint32_t)’:
/home/hankbillings/goestools/src/goesrecv/rtlsdr_source.cc:81:8: warning: unused variable ‘rv’ [-Wunused-variable]
   auto rv = rtlsdr_set_sample_rate(dev_, rate);
        ^~
/home/hankbillings/goestools/src/goesrecv/rtlsdr_source.cc: In member function ‘void RTLSDR::setTunerGain(int)’:
/home/hankbillings/goestools/src/goesrecv/rtlsdr_source.cc:91:7: warning: variable ‘rv’ set but not used [-Wunused-but-set-variable]
   int rv;
       ^~
/home/hankbillings/goestools/src/goesrecv/rtlsdr_source.cc: In member function ‘void RTLSDR::setBiasTee(bool)’:
/home/hankbillings/goestools/src/goesrecv/rtlsdr_source.cc:110:49: error: ‘rtlsdr_set_bias_tee’ was not declared in this scope
   auto rv = rtlsdr_set_bias_tee(dev_, on ? 1 : 0);
                                                 ^
/home/hankbillings/goestools/src/goesrecv/rtlsdr_source.cc: In member function ‘virtual void RTLSDR::stop()’:
/home/hankbillings/goestools/src/goesrecv/rtlsdr_source.cc:135:8: warning: unused variable ‘rv’ [-Wunused-variable]
   auto rv = rtlsdr_cancel_async(dev_);
        ^~
src/goesrecv/CMakeFiles/rtlsdr_source.dir/build.make:62: recipe for target 'src/goesrecv/CMakeFiles/rtlsdr_source.dir/rtlsdr_source.cc.o' failed
make[2]: *** [src/goesrecv/CMakeFiles/rtlsdr_source.dir/rtlsdr_source.cc.o] Error 1
CMakeFiles/Makefile2:2246: recipe for target 'src/goesrecv/CMakeFiles/rtlsdr_source.dir/all' failed
make[1]: *** [src/goesrecv/CMakeFiles/rtlsdr_source.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

hankbillings, 2018-07-10:
I'm using the raspbian image that was updated on 2018-06-27.  I guess i should go to the previous image and see if it works on that one?

pietern, 2018-07-10:
That function (`rtlsdr_set_bias_tee`) must be newer than the version shipping with Raspbian...

@ahobgood could you take a look? Perhaps mask this out if the librtlsdr version is too low?

hankbillings, 2018-07-11:
Is there a specific branch of rtlsdr that has rtlsdr_set_bias_tee? I'll install it if that's the solution.

hankbillings, 2018-07-11:
would the change @ahobgood made to 11 days ago from int to bool have anything to do with the issue? Just curious. Me not smart coder...

ahobgood, 2018-07-11:
@hankbillings the rtlsdr_set_bias_tee() function entered the osmocom/rtl-sdr trunk in version 0.5.4, while it appears that Raspbian currently ships v0.5.3 .  This was an oversight on my part when adding the patch.

Until we can get a build patch committed that properly detects if the function is available or not, you can certainly install a newer version of rtl-sdr libs if you require bias tee support.

Another option, if you do not need the bias tee, is to simply comment out line 110 in goestools/src/goesrecv/rtlsdr_source.cc and run 'make' again.

pietern, 2018-07-11:
Thanks for the analysis @ahobgood. With the latest commit it now uses pkgconfig to find librtlsdr and with it picks up the version that was installed. Only if the version is >= 0.5.4 will it allow using the bias tee.

closed 2018-07-11 by pietern

hankbillings, 2018-07-11:
Thank you. I appreciate you guys knocking this out.

commit 0440ae0c (2018-07-11): Use RTLSDR bias tee only if librtlsdr is new enough

Fixes #19.