pietern/goestools #28: Expose M1 location info

eminence, 2018-11-05:
As I understand it, the M1 payloads contain info about the area that GOES-16 is looking at.  This info is how goesproc is able to correctly project the map overlay.  Can this data be exposed to the user somehow?  (Perhaps via exif info in the images or something).  This would be useful for an end-user who wants to scan a bunch of images and find ones that are looking at particular areas of the globe.

pietern, 2018-11-11:
Great point. The first thing that comes to mind is optionally writing a separate JSON file for every image file that contains all the LRIT headers verbatim. EXIF is not an option because OpenCV doesn't support it. Also EXIF would only apply to JPEG files and not PNGs, PGMs, etc.

eminence, 2018-11-11:
a JSON file per image sounds perfect, too!

closed 2018-12-03 by pietern

eminence, 2019-05-02:
Following up on this after a long winter hiatus.  This new feature is working great, thank you!   Using the `ImageNavagation` header I can run my own projections to figure out what cities and states are included in an image (and thank you for the well-written comment about the projection in map_drawer.cc).

I'm curious about the `Table` structure.  Can you point me in the direction of some documentation that might describe what this table is used for?

pietern, 2019-05-05:
Thanks for the feedback. Glad to hear it's working for you.

Can you share a demo of where/how you're using it? 

The `Table` in the `ImageDataFunction` is a mapping of the 8 bit integer pixel values to the values in the underlying measurement system. It depends on the channel what you're looking at. These values are what's used for the gradient mapping, so you can express the colors gradient in Kelvin, and not worry how they are mapped to the pixel values.

pietern, 2019-05-05:
You can search for "image data function" in the LRIT specs (for example [here](https://www.noaasis.noaa.gov/LRIT/pdf-files/3_LRIT_Receiver-specs.pdf)) for more info.

eminence, 2019-05-08:
Thanks for the tips about the image data function info, I'll take a look

> Can you share a demo of where/how you're using it?

Sure!  It's being used for a personal dashboard I wrote to give me a quick interface to the images produced by goesproc.  https://goes.em32.net/  In the "Latest GOES-16 section" you can see that some images are labeled with some of the cites in view, and this was calculated using the ImageNavigation data from the json files

commit 9ad66afd (2018-12-03): Allow writing LRIT header as JSON file in goesproc

This uses the basename of the generated file as the basename of the
JSON file. The JSON file includes the path to the generated file under
the "Path" key. This may be a relative path, as it reuses the --out
option passed to goesproc.

Closes #28.