pietern/goestools #31: GOES 15 map is off

kidshare, 2018-12-11:
GOES 15 has moved to 128.57W and the map is off. I am receiving the GOES 15 images from GOES 17 relay. Everything else is working perfectly and I'm extremely pleased, thanks a bunch, I haven't had this much fun since I first tried wxtoimg several years ago!

[image]

pietern, 2018-12-18:
Thanks for reporting. I noticed this as well and hoped that the ground station HRIT feed assembly would be updated to fix this. With that not happening, the fix was easy. The override is in effect if the image was taken after Nov 7 and still includes a header with 135W as the longitude.

You can update to the latest version for the fix.

pietern, 2018-12-18:
And great to hear you're having fun with it :+1:

commit 11940d83 (2018-12-18): Override incorrect GOES-15 longitude

Fixes #31.
closed 2018-12-18 by pietern