pietern/goestools #32: Raspberry Pi Zero W

dmcvin2447, 2018-12-28:
Sorry if I am putting this in the wrong place.  New at participating in this.  I really enjoyed getting this to work on a Pi3B+ successfully running under Raspian Stretch Lite.

I tried to duplicate the process of making it work on a Raspberry Pi Zero W.  But, I keep on getting "Illegal instruction" error, when I execute the 'goesrecv' command as 'goesrecv -v -i 1 -c ~/goesrecv.conf'.  Works fine on the Pi3B+.

I was wondering if this will ever be made (or is even possible to be made) to work on a Raspberry Pi Zero W?

Thank you.

pietern, 2018-12-29:
Hi! Glad you're having fun with it so far. This is the right place, as is the mailing list which you can find at https://groups.google.com/forum/#!managemembers/goestools-users/.

The Pi Zero uses an older ARM CPU that doesn't include the NEON instruction set. These instructions are used to speed up the DSP routines in goesrecv. Without these instructions, it would run much slower. As is, goesrecv can already use more than 1 core on the Pi 3, so for the Pi Zero with only a single core, even if it had the same processor as the Pi 3, it wouldn't work. The Pi Zero is just too underpowered for this to work.

Cheers!

closed 2018-12-29 by pietern

dmcvin2447, 2018-12-29:
Thank you for the explanation.  Joined the Google group.