pietern/goestools #33: Issue with Assertion..

BLB5000, 2018-12-28:
Hi my issue is when I try to run goesrecv on command with my Airspy Mini & the usual command I'd ran it before: goesrecv -v -i 1 -c ~/goesrecv.conf it throws out this instance/error at me: 'std::runtime_error what(): Assertion 'rv >= 0' failed at /home/blake/goestools/src/goesrecv/airspy_source.cc:62 (I checked the line but didn't know how to fix it). Could somebody please tell me why this is & possibly how to fix it. Thanks for your time too hey..

pietern, 2018-12-29:
Hi!

This is odd... the routine sets the sample rate and getting a return value < 0 means it didn't succeed. This itself is odd because the sample rate value from the configuration file is validated in: https://github.com/pietern/goestools/blob/11940d83b8b3357a96f6782924e8b7a7096ab222/src/goesrecv/source.cc#L26-L37.

Can you confirm if you're running the latest version of the code? And post the contents of your `goesrecv.conf` file? Then we can take it from there...

pietern, 2019-05-08:
No response. Closing the issue.

closed 2019-05-08 by pietern

KR0SIV, 2019-07-12:
I'm getting this Assertion error as well.

`terminate called after throwing an instance of 'std::runtime_error'
   what():   Assertion  'rv >=0' failed at /home/corrosive/goestools/src/goesrecv/airspy_source.cc:62
Aborted (core dumped)`

I have the same issue with airspy and rtlsdr, I am however running within a xubuntu VM.
perhaps the VM is the issue.

I compiled with the current source and used the following for install and config: https://pietern.github.io/goestools/installation.html

KR0SIV, 2019-07-22:
Follow up: Yeah it was just my VM everything works fine on a native environment.