pietern/goestools #34: make -j2 install fails

dhacks541, 2019-01-06:
I've been following the very clear installation instructions for goestools installation, however when I run "sudo make -j2 install' it give the following errors:

[ 65%] Linking CXX executable areadump
[ 65%] Built target unzip
[ 65%] Linking CXX executable lritdump
CMakeFiles/areadump.dir/areadump.cc.o: In function `lrit::SegmentIdentificationHeader lrit::getHeader<lrit::SegmentIdentificationHeader>(std::vector<unsigned char, std::allocator<unsigned char> > const&, std::map<int, int, std::less<int>, std::allocator<std::pair<int const, int> > > const&)':
/home/pi/goestools/src/lrit/lrit.h:166: undefined reference to `lrit::SegmentIdentificationHeader lrit::getHeader<lrit::SegmentIdentificationHeader>(std::vector<unsigned char, std::allocator<unsigned char> > const&, int)'

....(end of error message):

collect2: error: ld returned 1 exit status
src/lrit/CMakeFiles/lritdump.dir/build.make:96: recipe for target 'src/lrit/lritdump' failed
make[2]: *** [src/lrit/lritdump] Error 1
CMakeFiles/Makefile2:1836: recipe for target 'src/lrit/CMakeFiles/lritdump.dir/all' failed
make[1]: *** [src/lrit/CMakeFiles/lritdump.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

closed 2019-01-06 by dhacks541

pietern, 2019-01-06:
@dhacks541 Did you figure out the issue?

dhacks541, 2019-01-06:
Not exactly, it was fixed by deleting everything and restarting the install. Not sure what went wrong. Sorry, should have tried such a simple fix before asking!