pietern/goestools #35: GOES 17 M1 time stamps

kidshare, 2019-03-19:
Not sure if there is a fix for this. M1 time stamps for channel 13 and channel 2 are now off by one minute and I'm not sure if it is coming from the satellite or the software. As a result, full color images are no longer generated. This is for M1 only, FD and everything else is fine. It was OK up until March 5 and briefly OK on March 13. For instance

Writing ./goes17/m1/ch13/2019-03-19/GOES17_M1_CH13_20190319T**1351**26Z.jpg (took 0.050s)
Writing: ./goes17/m1/ch02/2019-03-19/GOES17_M1_CH02_20190319T**1352**26Z.jpg (took 0.076s)

Is there a workaround?

pietern, 2019-03-19:
There isn't, unfortunately. The HRIT stream is assembled at a NOAA ground station and this software has had some timing related issues, where if the timing is just a tiny bit off, it will pick up a newer file for different channels. I'll ping Seth at NOAA about this to let him know this is happening.

closed 2019-03-19 by pietern

pietern, 2019-03-19:
Seth responded that he has reset the scheduler in the stream assembler.

I can confirm this is now working again starting with `GOES17_M1_FC_20190319T160727Z.jpg`.

kidshare, 2019-03-20:
Yes it is! Thank you very much!