pietern/goestools #37: Complete signal loss after a few minutes of use.

njbrown09, 2019-04-05:
When i turn it on after not using it for a bit, i get perfect signal (300 vit, 0 dropped packets). After around 5 minutes though, the vit goes up to 2200 and i get all dropped packets. Is this something normal for GOES or is this an issue with my setup/goestools? Im using Raspberry pi 2, RTLSDR (80 gain), and a GOES SAWBird powered via bias tee.

pietern, 2019-04-05:
Are you using an SDR with the E4000 tuner by any chance? I have had similar issues with mine in the past, and stopped using them. I recall reading about thermal issues with the on board LNA on that device. You can try reducing the gain until you lose signal and keep the gain at the lowest possible value. Otherwise, try a different SDR, ideally with a different tuner. I've never had problems with the R820T units.

njbrown09, 2019-04-05:
How do i check? I have a RTL-SDR Blog Device.

pietern, 2019-04-05:
When you start goesrecv it will log a message saying something along the lines of "Found tuner XYZ" or something similar. I think the ones from the blog are all R820T's though.

njbrown09, 2019-04-05:
Yes it does have a R820T in it. I just checked. Heres the console right when it starts. https://gyazo.com/499680f27d4a0d1afb0319a4fea4472e

pietern, 2019-04-05:
Another possibility is that your RPi2 is overheating and gets throttled. I have used an RPi2 in the past, with success, but at 95% or so CPU usage. If it overruns after a while it is possible it doesn't recover. If this is what's happening you should try using a RPi3.

njbrown09, 2019-04-06:
Tried it with a pi 3, it still had the same issue. I ordered a NooElec smart xtr and it will come tomorrow. Its definitely a problem with rtlsdr, as i have tried 2 of them and a rpi3.

njbrown09, 2019-04-06:
Solved. This was entirely due to the RTL-SDR dongle overheating.

closed 2019-04-06 by njbrown09

pietern, 2019-04-09:
Glad you were able to figure it out.