pietern/goestools #39: compile error - "proj version 4 required"

ilikeleptons, 2019-04-09:
I just pulled the latest version of the sources of goestools. When I try to compile goesproc (I believe) I get an error stating that "proj version 4 [is] required". Let me know what additional information you'll need, I'm a little new to this.

Here is the output from when I make the project:

`Scanning dependencies of target zip
Scanning dependencies of target version
Scanning dependencies of target util
[  0%] Building CXX object src/lib/CMakeFiles/zip.dir/zip.cc.o
[  0%] Building CXX object src/util/CMakeFiles/util.dir/fs.cc.o
[  0%] Building CXX object src/lib/CMakeFiles/version.dir/version.cc.o
[  1%] Building CXX object src/util/CMakeFiles/util.dir/string.cc.o
[  2%] Linking CXX static library libversion.a
[  2%] Built target version
Scanning dependencies of target dir
[  3%] Building CXX object src/lib/CMakeFiles/dir.dir/dir.cc.o
[  4%] Linking CXX static library libzip.a
[  4%] Built target zip
Scanning dependencies of target nanomsg
[  4%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/ep.c.o
[  4%] Linking CXX static library libdir.a
[  4%] Built target dir
Scanning dependencies of target timer
[  5%] Building CXX object src/lib/CMakeFiles/timer.dir/timer.cc.o
[  5%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/global.c.o
[  5%] Linking CXX static library libtimer.a
[  5%] Built target timer
[  5%] Building CXX object src/util/CMakeFiles/util.dir/time.cc.o
Scanning dependencies of target dcs
[  5%] Building CXX object src/dcs/CMakeFiles/dcs.dir/dcs.cc.o
[  5%] Linking CXX static library libutil.a
[  5%] Built target util
Scanning dependencies of target correct-reed-solomon
[  7%] Building C object vendor/libcorrect/src/reed-solomon/CMakeFiles/correct-reed-solomon.dir/polynomial.c.o
[  8%] Building C object vendor/libcorrect/src/reed-solomon/CMakeFiles/correct-reed-solomon.dir/reed-solomon.c.o
[  8%] Linking CXX static library libdcs.a
[  8%] Built target dcs
Scanning dependencies of target correct-convolutional
[  8%] Building C object vendor/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/bit.c.o
[  8%] Building C object vendor/libcorrect/src/reed-solomon/CMakeFiles/correct-reed-solomon.dir/encode.c.o
[  9%] Building C object vendor/libcorrect/src/reed-solomon/CMakeFiles/correct-reed-solomon.dir/decode.c.o
[ 10%] Building C object vendor/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/metric.c.o
[ 10%] Building C object vendor/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/history_buffer.c.o
/home/isaac/.cache/yay/goestools-git/src/goestools/vendor/libcorrect/src/convolutional/history_buffer.c: In function ‘history_buffer_search’:
/home/isaac/.cache/yay/goestools-git/src/goestools/vendor/libcorrect/src/convolutional/history_buffer.c:57:12: warning: ‘bestpath’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     return bestpath;
            ^~~~~~~~
[ 10%] Building C object vendor/libcorrect/src/convolutional/CMakeFiles/correct-convolutional.dir/error_buffer.c.o
[ 10%] Built target correct-reed-solomon
Scanning dependencies of target correct-convolutional-sse
[ 11%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/pipe.c.o
[ 11%] Building C object vendor/libcorrect/src/convolutional/sse/CMakeFiles/correct-convolutional-sse.dir/lookup.c.o
[ 11%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/poll.c.o
[ 11%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/sock.c.o
[ 13%] Building C object vendor/libcorrect/src/convolutional/sse/CMakeFiles/correct-convolutional-sse.dir/convolutional.c.o
[ 14%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/sockbase.c.o
[ 14%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/symbol.c.o
[ 14%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/ctx.c.o
[ 15%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/fsm.c.o
[ 15%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/pool.c.o
[ 15%] Building C object vendor/libcorrect/src/convolutional/sse/CMakeFiles/correct-convolutional-sse.dir/encode.c.o
[ 16%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/timer.c.o
[ 16%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/timerset.c.o
[ 16%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/usock.c.o
[ 16%] Building C object vendor/libcorrect/src/convolutional/sse/CMakeFiles/correct-convolutional-sse.dir/decode.c.o
[ 17%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/worker.c.o
[ 17%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/alloc.c.o
[ 17%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/atomic.c.o
[ 19%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/chunk.c.o
[ 19%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/chunkref.c.o
[ 19%] Built target correct-convolutional-sse
[... 232 lines cut ...]
[ 90%] Built target benchmark
[ 91%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/handler_goesr.cc.o
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:154: src/goesproc/CMakeFiles/goesproc.dir/handler_goesn.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
Scanning dependencies of target goesrecv
[ 91%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/goesrecv.cc.o
In file included from /home/isaac/.cache/yay/goestools-git/src/goestools/src/goesproc/map_drawer.h:5,
                 from /home/isaac/.cache/yay/goestools-git/src/goestools/src/goesproc/handler_goesr.cc:14:
/home/isaac/.cache/yay/goestools-git/src/goestools/src/goesproc/proj.h:4:2: error: #error "proj version 4 required"
 #error "proj version 4 required"
  ^~~~~
[ 92%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/config.cc.o
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:167: src/goesproc/CMakeFiles/goesproc.dir/handler_goesr.cc.o] Error 1
[ 92%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/options.cc.o
[ 94%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/decoder.cc.o
[ 94%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/demodulator.cc.o
[ 94%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/monitor.cc.o
make[1]: *** [CMakeFiles/Makefile2:2950: src/goesproc/CMakeFiles/goesproc.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 95%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/datagram_socket.cc.o
[ 95%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/source.cc.o
[ 95%] Linking CXX executable goesrecv
[ 95%] Built target goesrecv
make: *** [Makefile:152: all] Error 2
`

pietern, 2019-04-09:
Thanks for reporting the issue!

The output you included was a bit verbose so I edited it down to the relevant part.

Can you post the output of `cmake` instead of the output of running `make`? That will include information on the actual version of `libproj` you have installed. The fact that you got past that stage implies that CMake was able to find some version of libproj, just not the right one.

Please also include the operating system you're using and any relevant version information.

ilikeleptons, 2019-04-17:
ah, i forgot to mention that before i was building it from the archlinux user repository. however, when i instead follow the build instructions in README.md, i get far less progress:

cd goestools                                                                ~
isaac@thnikdap> mkdir -p build                                                    ~/goestools
isaac@thnikdap> cd build                                                          ~/goestools
isaac@thnikdap> cmake ../ -DCMAKE_INSTALL_PREFIX=/usr/local                 ~/goestools/build
-- The CXX compiler identification is GNU 8.2.1
-- The C compiler identification is GNU 8.2.1
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /bin/cc
-- Check for working C compiler: /bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Warning at CMakeLists.txt:36 (find_package):
  By not providing "FindSanitizers.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Sanitizers", but CMake did not find one.

  Could not find a package configuration file provided by "Sanitizers" with
  any of the following names:

    SanitizersConfig.cmake
    sanitizers-config.cmake

  Add the installation prefix of "Sanitizers" to CMAKE_PREFIX_PATH or set
  "Sanitizers_DIR" to a directory containing one of the above files.  If
  "Sanitizers" provides a separate development package or SDK, be sure it has
  been installed.

-- Defaulting build type to 'RelWithDebInfo'
CMake Error at CMakeLists.txt:59 (add_subdirectory):
  The source directory

    /home/isaac/goestools/vendor/libcorrect

  does not contain a CMakeLists.txt file.

CMake Error at CMakeLists.txt:60 (add_subdirectory):
  The source directory

    /home/isaac/goestools/vendor/libaec

  does not contain a CMakeLists.txt file.

CMake Error at CMakeLists.txt:68 (add_subdirectory):
  The source directory

    /home/isaac/goestools/vendor/nanomsg

  does not contain a CMakeLists.txt file.

CMake Error at src/lrit/CMakeLists.txt:5 (add_sanitizers):
  Unknown CMake command "add_sanitizers".

-- Configuring incomplete, errors occurred!
See also "/home/isaac/goestools/build/CMakeFiles/CMakeOutput.log".

i've attached the cmake output log below:
[CMakeOutput.log](https://github.com/pietern/goestools/files/3087594/CMakeOutput.log)

eminence, 2019-04-17:
Perhaps the submodules are missing?   Try:

    git submodule init
    git submodule update

pietern, 2019-04-17:
@ilikeleptons This problem will be resolved by the submodule initialization like @eminence.

I just checked the Arch Linux repository for proj and it has [version 5.2](https://www.archlinux.org/packages/community/x86_64/proj/).

AlCutter, 2019-04-25:
I hit this problem too with Ubuntu 19.04, which also ships Proj 5.20.

While digging around I noticed that `proj_api.h` shipped with 5.20 is compatible with the proj4 API, so I've created a small PR (in #43 ) which allows compilation to proceed when either proj version 4 or 5 are detected.

This allows the build to successfully complete on my machine.
Hope it's helpful!

pietern, 2019-04-27:
Thank you for fixing this!

closed 2019-04-27 by pietern