pietern/goestools #41: Is GOES HRIT active?

hjf, 2019-04-18:
I don't understand if the HRIT service is the same as EMWIN. 

According to NOAA's website, at https://www.nws.noaa.gov/emwin/ the HRIT/EMWIN services are "Not Operational".  

Does this mean I can't get images from GOES satellites now? Or is EMWIN unrelated to images that can be decoded by this software?

pietern, 2019-04-18:
The HRIT signal contains the EMWIN data stream. In the GOES-N days the EMWIN data stream was available both as its own narrow band signal as well as embedded in the LRIT stream. With the arrival of GOES-R (GOES-16 and GOES-17) the dedicated EMWIN signal is no longer present and the EMWIN data stream is only available through the HRIT signal.

They say it's not "operational" because they aren't ready to commit to the data stream being present reliably (due to all kinds of ground station data stream assembly systems that need to be qualified AFAICT).

The ABI images from the GOES satellites are separate from the EMWIN discussion. You can view EMWIN as a small subset of what you can decode with goestools.

pietern, 2019-04-18:
I recommend subscribing to and posting questions to the mailing list ([goestools-users](https://groups.google.com/forum/m/#!forum/goestools-users)), since it is monitored by many more people than just myself.

closed 2019-04-18 by pietern