pietern/goestools #48: channel numbering issues

eminence, 2019-05-16:
Following up to the comments in 96dc5544290e6f2742564a8795b960103a09a513 with an issue for better visibility.

`num` can be zero, which is tripping up the assertion.  Here's the whole `nlh` structure:

```
(gdb) print num
$1 = 0
(gdb) print nlh
$2 = {static CODE = 129, headerType = 129 '\201', headerLength = 14, agencySignature = "NOAA", productID = 16, productSubID = 0, parameter = 0,
  noaaSpecificCompression = 1 '\001'}
(gdb) print text
$3 = "Time of frame start = 2019-05-16T04:54:38.4Z;Satellite = G16;Instrument = FM1;Channel = N/A;Imaging Mode = ABI Mode 6;Region = Mesoscale;Resolution = 0.5km at nadir;Segmented = no"
```

pietern, 2019-05-16:
Thank you. Can you confirm this is not an issue for the full disks?

pietern, 2019-05-16:
I had a series of packets that didn't contain meso's and I assumed it would be the same for them.

eminence, 2019-05-16:
yes, it seems the full discs are fine.  in a different debugging session, i also extracted the lrit filename as `OR_ABI-L2-CMIPM1-M6C02_G16_s20191360506385_e20191360506442_c20191360506509.lrit` which is channel 02 of M1 i think

pietern, 2019-05-16:
Added another commit that falls back on extracting the channel from the filename.

eminence, 2019-05-16:
Cool, looking good.  I've successfully downloaded the full set of products (goes16 fd/m1/m2, plus goes17/fd).  

But I've also noticed that the channel name is again being set to an integer, and not "N/A".    

The last file that I downloaded with a "N/A" channel was `OR_ABI-L2-CMIPF-M6C13_G16_s20191361230342_e20191361240062_c20191361240137.lrit` and the first image that had a valid channel name was `OR_ABI-L2-CMIPF-M6C08_G16_s20191361230342_e20191361240050_c20191361240127.lrit`.  So maybe this was a temporary ground station glitch?

pietern, 2019-05-16:
Great, thanks for the confirmation.

I pinged Seth at NOAA about this issue and it turns out this was caused by a GOES-16 failover to a backup system that was already running a newer version of the PDA software. He then switched back to the primary running an older version which made the `Channel = N/A` disappear again.

Also see the big thread on the goestools-users mailing list [here](https://groups.google.com/d/msgid/goestools-users/be77f874-1a2b-452a-8c5b-057963716cbd%40googlegroups.com?).

eminence, 2019-05-16:
Good info, thanks.   I had seen the goestools-users mailing list, but struggled to participate.  In that thread you linked there are 2 messages that are marked as "This message has been deleted" -- that was me trying to post.    Do you see anything on your end about this?

Either way, this issue now seems pretty well solved, so I'll close this.

closed 2019-05-16 by eminence

pietern, 2019-05-16:
Hmm -- I didn't see those messages. I haven't heard of issues with this mailing list before.

Did your subscription go through?