pietern/goestools #49: Statsd monitoring for goesproc

bab5871, 2019-05-16:
Would be useful to have some statsd output from goesproc similar to goesrecv for monitoring purposes. I'm currently monitoring goesrecv statsd using telegraf/influxdb/grafana to plot all receiver stats and be alerted at anomalies in drops/vit errors. 

Currently there is no good way to monitor if goesproc stops working or has errors other than lack of image output or checking the console.

hemna, 2019-09-26:
Any hints on how to setup telegraf to slup the stats from goesrecv ?

elafargue, 2019-11-27:
This should give you a A to Z guide: https://github.com/elafargue/rpi-tig
The setup includes provisioning files for goestools

pietern, 2020-01-04:
@bab5871 Can you suggest what stats would be useful? Since you can configure arbitrary handlers, I'm thinking the stats should be a bit more low level, perhaps to count the number of LRIT files per type.

elafargue, 2020-01-06:
It would be nice to get the number of files per type, and maybe the size of the files?

bab5871, 2020-01-13:
I'd say maybe how many files per type we're processing. I also would be interested in seeing the number of drops or whatever other error information it spits out from time to time. I'll take a look when I get home and see if there's anything else I can think of.

creinemann, 2020-03-11:
After starting RPI-TIG I get this message/ error. Running on Pi4 with goestools running any thoughts? @elafargue elafargue

Starting TIG stack in the background
ERROR: Couldn't connect to Docker daemon at http+docker://localhost - is it running?

If it's at a non-standard location, specify the URL with the DOCKER_HOST environment variable.

creinemann, 2020-03-12:
> This should give you a A to Z guide: https://github.com/elafargue/rpi-tig
> The setup includes provisioning files for goestools

Is this assuming Docker is already installed on the pi?

elafargue, 2020-03-23:
Yes, you would have to install Docker first

mash55, 2020-07-08:
I am seriously questioning my cognitive status... I had the goes server reporting to my telegraf/influx/grafana server for months so at some stage I apparently was capable to figure out how set up everything. For reasons I don't wanna elaborate, I lost my TIG server and I am in process of setting everything up again... but for some reason I can get the GOES server back into reporting ton telegraf (except my scripts reporting temperatures and image collection). I would appreciate any help getting goesproc back o reporting o my TIG.
Thanks

mash55, 2020-07-09:
goesrecv.conf
  |__ [monitor]
      statsd_address = "udp4://host:8125"

In case somebody else is drawing a blank...