pietern/goestools #51: Not getting a signal

Hellmanx, 2019-05-28:
I have been trying a couple of weeks to try to get this working and I've had no luck, nor any luck from Reddit to get any help.

My issue is that i'm not getting packets or a good vit signal (above 2000)

But what confuses me is that, on windows, I have used [Satcom-decoder](https://usa-satcom.com/tag/software/) and have gotten an [image], with several pictures.  And the signal is perfectly visible [viewing it on SDR#](https://user-images.githubusercontent.com/51136027/58508378-e7aa0180-8161-11e9-8d28-5714c43fdf5c.png)

So, I'm not sure what the issue here is.

hemna, 2019-09-26:
try bumping the gain to 30 ? in the goesrecv.conf ?

I had a problem using an rpi where I was getting undervoltages.   I had to power the rpi from a real AC adapter and the LNA from a separate adapter.  Once I did that, then I got vit(avg) rates down to 300 and 0 droped packets

unixpunk, 2019-09-26:
I've experienced the same thing with a rpi3 - if the red power light flashes at all during boot or operation, even once, then you have an under-voltage issue.  There are also two different SAWbird LNA's for GOES reception, one is +20db and super-low power and the other is +30ish but uses too much power to be supplied by any SDR except the RTLSDRv3.  

I've found with the former that I need to drive the cable run with an additional LNA, in my case, an LNA4ALL so that I could bridge power from bias-t, out the LNA4ALL and to the + pin on the SAWbird.  With the latter LNA, I've not needed an additional LNA in the cable chain.  I'm using the 1.9ghz grid antenna with no modifications.  HTH