pietern/goestools #52: CONUS data now getting processed by goesproc

uuacallis, 2019-06-22:
Looking at https://www.ssec.wisc.edu/datacenter/satellite-schedules/goes-16-schedules/

It appears that every five minutes there should be a CONUS content downloaded, however I am not seeing that. 

I am seeing the M1 / M2 / FD regoins, but nothing more.

eminence, 2019-06-22:
I think this page is describing what NOAA can download from GOES, not what we can download with goesproc (which can only receive what NOAA processes at their base stations and uploads to the satellite for re-broadcast).

pietern, 2019-06-23:
Thanks, @eminence. Indeed the HRIT stream is a subset of the products listed on that page.

closed 2019-06-23 by pietern

uuacallis, 2019-06-24:
> Thanks, @eminence. Indeed the HRIT stream is a subset of the products listed on that page.

Thanks @pietern  - Is there a holistic list of what should be in the stream? 
I am even noticing that the FD images aren't coming as often as they used to, so I am presuming that either A - I am doing something wrong, or B - I am mis-interpreting it.

tmbates12, 2019-06-24:
Provided nothing goes awry, I get 48 images a day (every half hour) for the full disk images, as far as M1 and M2, depending on the day it can reach up to around 100 images. I've got a web server set up with the images if you want something to compare with.
https://goes.freakinbananas.xyz/

pietern, 2019-06-24:
@uuacallis Make sure you don't have any packet drops on the goesrecv side. If you do have packet drops then it is possible not all images will show up. Also keep an eye out on available disk space.