pietern/goestools #55: GoesProc fails to interpret frame dates correctly

whiten01se, 2019-07-04:
On certain (64bit?) Linux systems the function parseTime(std:string& in, struct timespec* ts) in time.cc fails to correctly interpret the input string to a valid time_t, and therefore produces an invalid timespec result (value -1, resulting in 1969-12-31T23:59:59 dates for everything).

The string is parsed into a tm struct (containing separate fields for year, month, day etc), which then calls **mktime()** to turn this into a time_t value.

The call to mktime() produces -1 on my Ubuntu 19 64bit system however, meaning none of the frame start timestamps are interpreted correctly.
On Raspberry Pi (latest Raspbian) this works fine.

I tried to separate the date conversion code into a standalone program, and interestingly here it works fine with the exact same code of parseTime(), with the exact same input that was observed to produce -1 in goesproc.

Somewhere along the execution of everything goesproc does, the behavior of mktime changes and no longer produces valid values.

whiten01se, 2019-07-04:
And i just stumbled across a solution for this, i'll make a PR for it.
The "trick" is to null out the tm variable with memset() before using it, to make sure that all fields are set to 0 before assigning any values to it. If that doesn't happen, other fields that we don't use, will have random values in them, causing the issues.

pietern, 2019-07-30:
Fixed in #56.

closed 2019-07-30 by pietern