pietern/goestools #61: terminate called after throwing an instance of 'std::runtime_error'

wds657, 2019-07-31:
Hi this is a error. I del all and reinstall it, and the same, 
try new goesproc-goesr.conf config reading here, adding and removing lines in goesproc-goesr.conf, and many times say console, invalid image handler product. someone have a "goesproc-goesr.conf" final please? 
I try all, the process crash after minutes
Thanks

i@GOES16:~ $ sudo goesproc --config /home/pi/goestools/goesproc-goesr.conf --mode packet --subscribe tcp://127.0.0.1:5004
Writing: ./nws/2019-07-31/20190731T230700Z_2019072122307548-pac48_latestBW.gif
Writing: ./nws/2019-07-31/20190731T230700Z_201907212230707-pacsfc72_latestBW.gif
terminate called after throwing an instance of 'std::runtime_error'
  what():  Assertion `rv == 2` failed at /home/pi/goestools/goestools/src/goesproc/handler_goesr.cc:29, Expected to extract mode and channel from file name (OR_ABI-L2-SSTF-M6_G16_s20192122200487_e20192122300196_c20192122306390.lrit)

tmbates12, 2019-07-31:
Run `git pull origin master` in your cloned goestools folder to update with the latest commit, this is a duplicate of #57

wds657, 2019-08-01:
I del all and install and the same, the config file goesproc-goesr.conf no decode image 
it seems to me and crash

pietern, 2019-08-01:
You're running an older version of goesproc. When you reinstall, what
commands do you run?

When you start it without specifying a path (i.e. just "goesproc") then it
will try to find one in /bin, /usr/bin, /usr/local/bin, etc. Perhaps you
didn't run "make install" when reinstalling?

Cheers,
Pieter

wds657, 2019-08-01:
this commands install
**************************************************************************
git clone https://github.com/pietern/goestools.git
cd goestools
git submodule init
git submodule update --recursive
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr ..

# this will take a while on a raspberry pi
sudo make -j2 install 
**************************************************************************
work, I see images perfect VISUAL, INFRARRED, CLOUD TOPS,  but error process always certain images products ACHTF, RRQPEF, SSTF ....

that's why I wanted the file goesproc-goesr.conf here is where I think is the problem

I install system Pi (Raspberry OS) on notebook intel i3 with 8gb ram and 250gb disc

wds657, 2019-08-05:
Hi I del all and reinstall it and work, except that compile goestools in other computer netbook, in the notebook say goesrecv "segment fault" ??, install all on netbook, raspberry desktop, compile goestools, and I took out disk and put on the other notebook and no error for now, 
Now I will try to modify the goes r conf for more products images and see what happens

mlinuxguy, 2020-02-03:
FYI,  I changed the gain to 49.6 from 5 (in goesrecv.conf) and got this error:
root@raspsdr:/etc# goesrecv -v -i 1 -c ~/goesrecv.conf
```
terminate` called after throwing an instance of 'std::runtime_error'
  what():  type error: this value is double but int was requested
Aborted
```
Silly mistake but documenting it here to help others.
It dies in:
```
[New Thread 0x76b96450 (LWP 6731)]
terminate called after throwing an instance of 'std::runtime_error'
  what():  type error: this value is double but int was requested

Thread 1 "goesrecv" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x76be0824 in __GI_abort () at abort.c:89
#2  0x76e13060 in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#3  0x76e10db0 in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0x76e10e24 in std::terminate() () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#5  0x76e11158 in __cxa_throw () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#6  0x0002b1b4 in toml::failwith<char const (&) [27], char const*, char const (&) [6], char const*, char const (&) [15]>(char const (&) [27], char const*&&, char const (&) [6], char const*&&, char const (&) [15]) (args#0=...,
    args#1=args#1@entry=<unknown type in /root/goestools/build/src/goesrecv/goesrecv, CU 0x21af4, DIE 0x366cc>,
    args#2=...,
    args#3=args#3@entry=<unknown type in /root/goestools/build/src/goesrecv/goesrecv, CU 0x21af4, DIE 0x366cc>,
    args#4=...) at /root/goestools/vendor/tinytoml/include/toml/toml.h:425
#7  0x00028614 in toml::Value::assureType<int> (this=0x9c328)
    at /root/goestools/vendor/tinytoml/include/toml/toml.h:1258
#8  toml::Value::ValueConverter<int>::to (this=<optimized out>, v=...)
    at /root/goestools/vendor/tinytoml/include/toml/toml.h:1185
#9  toml::Value::as<int> (this=0x9c328) at /root/goestools/vendor/tinytoml/include/toml/toml.h:1270
#10 (anonymous namespace)::loadRTLSDRSource (v=..., out=...) at /root/goestools/src/goesrecv/config.cc:178
#11 Config::load (
    file="\204\357\377~H\205\375v\001\000\000\000\034\236\333v\225\022\000\000\214\316\333v\230\245\377v\204\357\377~\200\357\377~\214\316\333v\360\354\377v\214\316\333v\314t\332v\332\026\207\060\266\070\204\001\373W\336v\034\236\333vL\310\376v\360\354\377v\360\357\377~\210\370\377vܱ\376v\230\245\377v\350\357\377~\334:\332v \a\000\000\354\357\377~H\205\375v\001\000\000\000\314\346\332v \a\000\000\214\316\333v\230\245\377v\354\357\377~\350\357\377~\373W\336v\360\354\377v\214\316\333v\314t\332v\303\321{)\216\336K\001\060\211\001\000\fo\001\000L\310\376v\360\354\377vX\360\377~\210\370\377vܱ\376v\000\360\377v\000\000\000\000"...) at /root/goestools/src/goesrecv/config.cc:391
#12 0x0002592c in main (argc=<optimized out>, argv=<optimized out>)
    at /root/goestools/src/goesrecv/goesrecv.cc:21

```