pietern/goestools #63: proj initialization error: major axis or radius = 0 or not given

btashton, 2019-08-11:
I'm seeing this quite frequently while I am receiving data from GOES-17.
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  proj initialization error: major axis or radius = 0 or not given
```

I'll try and hook gdb up to get some more context for the crash since I seem to be able to recreate it without too much issue.

cottsay, 2019-09-10:
I remember hitting something like this a while back.
Give this patch a shot: https://copr-dist-git.fedorainfracloud.org/cgit/cottsay/goestools/goestools.git/tree/goestools-proj-compat.patch

btashton, 2019-10-28:
Looks like that resolved the issue.  It is also not an issue on fedora 30 with the later libraries.

closed 2019-10-28 by btashton

pietern, 2020-01-04:
@btashton Do you recall which version of libproj caused the crash?

pietern, 2020-01-04:
@cottsay Thanks for posting your fix. It looks trivial, but could have side effects, so would like to know which version of libproj it needs to be applied for.

gvfd107, 2022-09-23:
i am running into this after installing anaconda, and satpy, im not sure if that helps. but i have never installed a patch, so im not sure how to patch this problem