pietern/goestools #67: VCDU and TP_PDU drops?

robman501a, 2019-10-22:
Hi there, 
So once and awhile I am seeing a 'VC 14: VCDU drop' and a VC 14: Detected TP_PDU drop' in the terminal window that goesproc is running in. 
The command I am using is....
goesproc -c /usr/local/share/goestools/goesproc-goesr.conf -m packet --subscribe tcp://127.0.0.1:5004

Goesproc-goesr.conf is saving the images to my NAS over a wired connection.

So, what is causing this error?

Both goesproc and goesrecv are running on my Raspberry Pi 3 B.

eminence, 2019-10-23:
Hi.  These messages indicates that the receiver (goesrecv) dropped some packets.  That is, the receiver wasn't able to decode/receive some data from the satellite.   For most of us, these are fairly common errors because few of us are receiving under ideal conditions.

If you have run the goesrecv command with the `-v` argument, you should see a continuous printout of the number of packets received and dropped.  The VCDU and TP_DPU drops from goesproc will coincide with a report of dropped packets from goesrecv.

robman501a, 2019-10-23:
Could the VCDU and TP_DPU drops be caused by if a cloud or an aircraft passes by and weakens the signal strength?
This is how I have the program running. 
```
goesrecv -v -i 1 -c /home/pi/goesrecv.conf
goesproc -c /usr/local/share/goestools/goesproc-goesr.conf -m packet --subscribe tcp://127.0.0.1:5004
```
As you can see in the attached image, everything is currently running well.
[image]
I think I'm starting to understand now.

eminence, 2019-10-23:
I don't think an aircraft would impact reception (I think it would be too quick to cause any real impact), but weather can definitely impact signal strength.  In my personal setup, rainy weather always causes problems (sometimes 100% packet loss)

pietern, 2020-01-04:
Thanks, @eminence.

@robman501a The screenshots you provide show a Viterbi error rate of close to 400 bit errors per packet. This is also the cause of the high number of Reed Solomon corrections that need to be applied. At about 500 bit errors per packet you start to see packets that can no longer be corrected and are dropped. When they are, you'll see the "drops" column in the goesrecv output show non-zero numbers, and you'll see goesproc complain about packet drops.

Some additional background. Every LRIT file is built up from many packets. A single TP_PDU packet is up to 8192 bytes large and can therefore span multiple VCDU packets. If it sees packet drop in the middle of decoding a TP_PDU packet, it will drop the whole thing. If it's the first packet in a sequence of TP_PDU packets, it will show the VCDU drop error.

closed 2020-01-04 by pietern