pietern/goestools #68: What can goestools do exactly?

robman501a, 2019-10-26:
Having no issues but I would like to know what this program does exactly besides produce images and text files from the NWS. Not sure if there's more but it's only giving me JPG files from GOES and GIF and TXT files from the NWS.
I was wondering about the following...
Geostationary Lightning Mapper (GLM),
Data Collection System (DCS),
Emergency Managers Weather Information Network (EMWIN),
Search and Rescue Satellite Aided Tracking (SARSAT),
GOES Rebroadcast (GRB)? Would that be the files in the GOES-xx directory?

Source:
https://en.wikipedia.org/wiki/GOES-16

Also, is there a GUI frontend for goesrecv?

n2cr, 2019-10-26:
EMWIN functionality can be enabled via the goesrecv config file.  GLM is 
not currently included in the HRIT data stream and is only available via GRB.

GRB requires different equipment and software.

robman501a, 2019-10-26:
How do you enable EMWIN?
Here is my goesrecv.conf file.
```
[demodulator]
mode = "hrit"
source = "rtlsdr"

[rtlsdr]
frequency = 1694100000
sample_rate = 2400000
gain = 30
bias_tee = true

[costas]
max_deviation = 200e3

[decoder.packet_publisher]
bind = "tcp://0.0.0.0:5004"
send_buffer = 1048576

[monitor]
statsd_address = "udp4://localhost:8125"
```

eminence, 2019-10-26:
The `goesrecv.conf` file is for the receiver (goesrecv, the program that demodulates and decodes the signal).  

To get EMWIN, you need to enable it in the processor (goesproc).  Click [here](https://github.com/pietern/goestools/blob/24c7fcb26dbf90257c7428f29dd28378dcc6bca0/etc/goesproc.conf#L69) for an example.

robman501a, 2019-10-26:
Okay, thank you. Any idea what other settings I should add to get the full potential of the GOES satellites and this program?

tmbates12, 2019-10-26:
SARSAT is on a separate transponder at around 1545MHz, and there's several
different programs that can decode the EPIRBs.
As far as DCS is concerned, there's the raw signals from base stations at
1680MHz, and there's the decoded data available in the HRIT stream,
however goestools doesn't have support for it as it stands; I'd work on
adding it if I wasn't so busy with school.

There's also support now for the Non-CMIP products on the HRIT downlink
now, they can be enabled in the goesproc config. Here's an example
of how you'd add these:
https://github.com/pietern/goestools/pull/59/commits/3e628fa38bf81c4380e79bdcf21921f396e096cb

robman501a, 2019-10-26:
Thanks! This is great stuff guys. Though, I am really surprised it isn't covered in the documentation.

robman501a, 2019-10-26:
Hey guys, I was just reading this. https://www.nws.noaa.gov/emwin/HRIT-EMWIN%20User%20Group_Final_4_18.pdf It says that images from Himawari come down on VCID 60 every 60min. If that's so, why aren't I getting any? Same with GOES-15 on VCID 6. It's like it doesn't want to give me and LRIT products. It did give me those a few days ago but that was it.

eminence, 2019-10-27:
Himawari and GOES-15 indeed are part of the rebroadcast, but not always.  I don't know the schedule, nor how NOAA decides when to include them.  For Himawari, I last received a full-disc on October 24 2019.  For GOES-15, I last received a full-disc on July 25th, but have received other images also on October 24th.

As for admin text, I haven't received any of those since August 13th.  I'm honestly not sure why.

robman501a, 2019-10-27:
Hmm. That is indeed odd. Would decoding LRIT be any benefit or should I just leave it on HRIT?

eminence, 2019-10-27:
Ah, I see what's going on with the admin messages.  They are still being broadcast, but with an old date:

    GOES_EAST_Admin_message_updated_2019.07.11.lrit

So I think goesproc is smart enough to not write it to disk again, since it already exists.

robman501a, 2019-10-28:
That might explain why I'm not getting Himawari and GOES-15 also. It might be the same reason.
How do you get the .lrit files anyway and how would you decode them?

pietern, 2020-01-04:
The missing Himawari and GOES-15 products was addressed in #71.

Use of LRIT vs HRIT is a historical thing. As far as I know, the LRIT **file format** and the LRIT **signal** were first developed for the GOES-N series (about 2 decades ago). For the GOES-R series satellites, the HRIT **signal** replaced the LRIT **signal**, but it continued to use the LRIT **file format**. This is at the root of some of the confusion. The goesrecv mode is configurable because when I originally wrote the code, both GOES-13 and GOES-15 were still operational and transmitting the LRIT signal. In the mean time, they have both been replaced by GOES-16 and GOES-17, respectively, and we only have to deal with the HRIT signal now. There is no longer an operational satellite that broadcasts the LRIT signal.

closed 2020-01-04 by pietern

creinemann, 2020-03-04:
> The `goesrecv.conf` file is for the receiver (goesrecv, the program that demodulates and decodes the signal).
> 
> To get EMWIN, you need to enable it in the processor (goesproc). Click [here](https://github.com/pietern/goestools/blob/24c7fcb26dbf90257c7428f29dd28378dcc6bca0/etc/goesproc.conf#L69) for an example.  I have looked at the example for EMWIN on goesproc.conf. but how does one "enable" it. It seems my setup, which is working well, only has goesrecv.conf, not the goesproc.conf (except the example version in ..\goestools\etc\.  What would be the command to run the goesproc to be able to receive EMWIN?

CLGPerformance, 2023-06-17:
> Having no issues but I would like to know what this program does exactly besides produce images and text files from the NWS. Not sure if there's more but it's only giving me JPG files from GOES and [...]

> 

https://www.noaasis.noaa.gov/GOES/GRB/grb.html

creinemann, 2023-06-17:
You won't get GLM, or GRB , those plus SUVI come in on GRB only and on Level 1 Data. EMWIN can be processed on goestools, along with NON CMIP imagery and DCS via HRIT with the correct handlers.
check my page for a ton of info [https://usradioguy.com/goes-satellite-imagery-reception/](https://usradioguy.com/goes-satellite-imagery-reception/)

For a front end, try Vitality GOES, highly configurable and displays everything you have set up to receive: https://usradioguy.com/satellites/vitality-goes-web-app/

[image]
[image]