pietern/goestools #69: How do I change the color?

robman501a, 2019-11-22:
How do I change the color of my full disk image to look like the one from  https://www.star.nesdis.noaa.gov ? Both are 5424 x 5424.
I find the color could be better.
[image]

robman501a, 2019-11-26:
Would it require a larger dish to capture channel 1?

eminence, 2019-11-27:
In theory, yes.  I believe that all ABI channels are available on the GOES rebroadcast downlink (GRB), but 1) you need a ~3m dish, and 2) different software.   The `goestools` software only does HRIT decoding.  

I found some additional information here:  https://www.teske.net.br/lucas/2017/10/goes-grb-first-light/

pietern, 2020-01-04:
That's right. The GeoColor composite requires channels 1, 2, and 13. On HRIT we only have 2 and 13.

closed 2020-01-04 by pietern