pietern/goestools #71: Missing: Himawari-8, GOES-16 and GOES-17 relays

robman501a, 2019-11-27:
I know that goestools does capture the images from these three satellites that do get relayed through GOES-16 because it's in the config file and I did get it once, a long time ago. How come they are not more constant?
Also, shouldn't I get an admin.txt file?

robman501a, 2019-11-27:
Just reading this...
> Also included in the HRIT/EMWIN broadcasts are graphic versions of Japan Meteorological Agency (JMA) Himawari and EUMETSAT Meteosat Second Generation Seviri data. NOAA receives Himawari visible, Infrared and Water Vapor images from the Japanese Meteorological Agency and converts them to NOAA HRIT format.
https://www.noaasis.noaa.gov/GOES/HRIT/products.html

Himawari (HBI) and Meteosat (MSG –Severi) is transmitted hourly,
GOES West / GOES-15 (GVAR) is every 3 hours
so why aren't I getting it?

mash55, 2019-12-11:
I am getting GOES15/16/17 & himawari with out of the box config file from GOES17

robman501a, 2019-12-11:
Okay, maybe since that is more focused on the west, I'm wondering if GOES-16 just doesn't send it out. I might have to change to GOES-17 for this.

pietern, 2020-01-04:
Thanks, @mash55. I no longer have direct access to the feeds myself (so I cannot confirm first hand), but this explanation sounds plausible. Closing the issue as there is no goestools work required.

closed 2020-01-04 by pietern