pietern/goestools #73: goesproc: std::invalid_argument -- what():  stoi

cswanson420, 2019-12-23:
Running goesproc for the last year in docker containers [ Ubuntu trusty, docker-engine 1.11.2-0~trusty --  3.19.0-25-generic #26~14.04.1-Ubuntu SMP Fri Jul 24 21:16:20 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux -- libc6:amd64 2.19-0ubuntu6.7 -- gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04)]. 

Today, I noticed this error in the docker logs:

{"log":"terminate called after throwing an instance of 'std::invalid_argument'\n","stream":"stderr","time":"2019-12-23T05:36:16.062325837Z"}
{"log":"  what():  stoi\n","stream":"stderr","time":"2019-12-23T05:36:16.06236406Z"}

This is in dmesg:

[46703.734555] traps: goesproc[30816] general protection ip:7f1b3be75196 sp:7ffed5bc3f90 error:0 in libc-2.23.so[7f1b3be3e000+1c0000]

I am running two independent instances of goesproc: one for goes16 and one for goes17 goesrecv instances. These goesrecv instances are running on rpi hosts which are connected directly to their respective antenna.

This is the key section for the docker image creation:

RUN \
  apt-get install -y \
     build-essential \
     cmake \
     git-core \
     libopencv-dev \
     zlib1g-dev

# download, compile, and install librtlsdr
RUN \
  cd /opt && \
  apt-get install -y libusb-1.0-0-dev && \
  git clone https://github.com/steve-m/librtlsdr.git && \
  cd librtlsdr && \
  mkdir build && \
  cd build && \
  cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr -DINSTALL_UDEV_RULES=ON .. && \
  make -j2 install

RUN \
  cd /opt && \
  git clone --recursive https://github.com/pietern/goestools && \
  mkdir -p /opt/goestools/build && \
  cd /opt/goestools/build && \
  cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local && \
  make && \
  make install

Both docker instances crash within minutes of each other which is odd and rarely longer than 30 minutes.

I am running this git hash version: commit 9ad66afda56c68100191b37d9d240a4b7a7a2d8f which has an stoi call in "if (key == "Channel")" control block. The other instance of stoi is in a try/catch. 

The latest master version has omitted the stoi function in the "if (key == "Channel")" control block. 

I am curious if the data has changed from the satellite transmission that is causing a fault in goesproc which could be causing this failure or could it be something else.

cswanson420, 2019-12-23:
This seems to have been solved by recompiling goestools from master (as of 20191223) as the process has been running for 4x the time it would have crashed in. 

Hopefully this helps somebody else.

closed 2019-12-23 by cswanson420

pietern, 2020-01-04:
Thanks for the follow up, @cswanson420. This issue started happening on May 16 last year and was reported on [the mailing list](https://groups.google.com/d/msg/goestools-users/UgFkYTthKzM/xOUGS4GsBAAJ) and fixed on the same day. See #48 for a few related issues.