pietern/goestools #75: Build failure of goesrecv. Rpi 3B+ Ubuntu Server 18.04 64 bit

jtsfour, 2020-03-27:
Building of goestools keeps failing on my Raspberry pi 3 B+ running Ubuntu Server 18.04 64 bit.

After running make the compiler hangs and crashes on.
'[90%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/monitor.cc.o'

Here is a pic of where the crash occurs
[image]

This is the first part of the cmake output
[image]

This is the second part of the cmake output
[image]

If you want i can email an archive of my build directory.  Its too large for github.

I have tried rebooting.  I have also tried re-downloading and rebuilding 2-3 times. The result is always the same.  I recently built the same software yesterday on Linux Mint (on amd64 not arm) and had no notable issues.

I have been following these instructions.  https://gist.github.com/lxe/c1756ca659c3b78414149a3ea723eae2#file-goes16-rtlsdr-md

eminence, 2020-03-29:
The `Killed (program cc1plus)` makes me think this is an out-of-memory issue, and the kernel has killed the compiler for using too much memory.  You should be able to confirm this by looking at the logs in either the command `dmesg` or in the log files in `/var/log/messages`.

jtsfour, 2020-03-29:
@eminence Yep you are right.  The fix to this 'issue' is to run make in a single thread.  To anyone attempting to run this configuration make sure you do not use the 'j' flag for make.

Run 
make install

Not 
make -j2 install

closed 2020-03-29 by jtsfour