pietern/goestools #77: std::runtime_error - Expected signature 0x06054b50, got signature 0x2e4b4c54

nigelvh, 2020-05-01:
I've gotten a couple of these in the last couple days. I ignored the first as a one-off, but since it's repeated, here I am. A couple of sample outputs:

```terminate called after throwing an instance of 'std::runtime_error'
  what():  Expected signature 0x06054b50, got signature 0x2e4e5447
Aborted (core dumped)
```

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Expected signature 0x06054b50, got signature 0x2e4b4c54
Aborted (core dumped)
```

Not sure where in the code this is sourcing from. Is there a good option for additional debug logging that would help, or does someone have an idea for the source of this issue?

nigelvh, 2020-05-04:
Another instance today:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Expected signature 0x06054b50, got signature 0x2e495758
Aborted (core dumped)
```
Seems like the expected signature is always the same, but the "got" signature is changing.

pietern, 2020-05-05:
Thanks for reporting.

The error is produced by code that interprets a zip file. Only the EMWIN data stream includes zip files, so some file in that stream must be triggering the problem.

Crashing the whole process is clearly unnecessary. It should simply log a warning and ignore the bad file. I doubt you'd see those warnings given the amount of output goesproc produces, but that's a different issue. I just pushed a commit that implements this. Can you give it a try?

nigelvh, 2020-06-03:
Apologies for the delay. I've compiled and am running the new version. The issue was relatively infrequent to begin with, but I'll update if there's any issues.