pietern/goestools #90: GoesTools "No package 'opencv' found"

An0nymo0se, 2020-06-12:
I tried installing goestools, but on every single occasion, goestools cannot find opencv. To verify that I actually had opencv installed, I ran sudo dpkg -l and found that opencv 4.2.0 (and it's associated libraries) were installed. I've also tried installing and building opencv 2.4.13.6, which gave the same results.

An0nymo0se, 2020-06-12:
The full readout:

-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Defaulting build type to 'RelWithDebInfo'
-- Looking for dotprod in FEC
-- Looking for dotprod in FEC - not found
-- Performing Test HAVE_SSE
-- Performing Test HAVE_SSE - Success
-- Looking for include file malloc.h
-- Looking for include file malloc.h - found
-- Looking for include file stdint.h
-- Looking for include file stdint.h - found
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Performing Test HAVE_DECL___BUILTIN_CLZLL
-- Performing Test HAVE_DECL___BUILTIN_CLZLL - Success
-- Inline keyword found - inline
-- Restrict keyword found - restrict
-- Detected nanomsg ABI v5.1.0
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- OS System is Linux
-- OS Version is 5.4.0-37-generic
-- Looking for gethrtime
-- Looking for gethrtime - not found
-- Looking for socketpair
-- Looking for socketpair - found
-- Looking for eventfd
-- Looking for eventfd - found
-- Looking for pipe
-- Looking for pipe - found
-- Looking for pipe2
-- Looking for pipe2 - found
-- Looking for accept4
-- Looking for accept4 - found
-- Looking for epoll_create
-- Looking for epoll_create - found
-- Looking for kqueue
-- Looking for kqueue - not found
-- Looking for poll
-- Looking for poll - found
-- Looking for getaddrinfo_a in anl
-- Looking for getaddrinfo_a in anl - found
-- Looking for clock_gettime in rt
-- Looking for clock_gettime in rt - found
-- Looking for sem_wait in rt
-- Looking for sem_wait in rt - not found
-- Looking for sem_wait in pthread
-- Looking for sem_wait in pthread - found
-- Looking for gethostbyname in nsl
-- Looking for gethostbyname in nsl - found
-- Looking for socket in socket
-- Looking for socket in socket - not found
-- Looking for CLOCK_MONOTONIC
-- Looking for CLOCK_MONOTONIC - not found
-- Looking for atomic_cas_32
-- Looking for atomic_cas_32 - not found
-- Looking for AF_UNIX
-- Looking for AF_UNIX - found
-- Looking for backtrace_symbols_fd
-- Looking for backtrace_symbols_fd - found
-- Performing Test NN_HAVE_MSG_CONTROL
-- Performing Test NN_HAVE_MSG_CONTROL - Success
-- Performing Test NN_HAVE_GCC_ATOMIC_BUILTINS
-- Performing Test NN_HAVE_GCC_ATOMIC_BUILTINS - Success
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libairspy'
--   Found libairspy, version 1.0
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.6.0
-- Checking for module 'opencv'
--   No package 'opencv' found
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:643 (_pkg_check_modules_internal)
  src/goesproc/CMakeLists.txt:23 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/username/goestools/build/CMakeFiles/CMakeOutput.log".
See also "/home/username/goestools/build/CMakeFiles/CMakeError.log".

Chaemelion, 2020-08-29:
I bypassed this issue by editing src/goesproc/CMakeLists.txt and changing `pkg_check_modules(OPENCV REQUIRED opencv)` into `pkg_check_modules(OPENCV REQUIRED opencv4)`. Then I added `include_directories(BEFORE SYSTEM /usr/include/opencv4)` to line 84 of the main CMakeLists.txt file. Of course after all this I ran into another problem with the version of proj, but pull request #83 solves that issue for now. 

I've now got everything compiled on a brand new install of Ubuntu 20.04.1, just waiting on my antenna now.

pietern, 2022-02-13:
This should be fixed with 6440bbbf0337a062ac8400fb4ccc32ffffac98bb.

Confirmed to work on Ubuntu 20.04, 21.04, and 21.10.

closed 2022-02-13 by pietern