pietern/goestools #91: M_SDU continuation failed

nigelvh, 2020-07-20:
Seeing these errors today in goesproc: `VC 8: M_SDU continuation failed; 34931 byte(s) needed to complete M_SDU, 188 byte(s) available`

Looks like it got stuck there and kept repeating that message. Killing/restarting appears to be running OK at the moment.

Currently running version 543ece3dc99ac865d87ac03490be99cf1f0457a7 from the testing in https://github.com/pietern/goestools/issues/77

pietern, 2020-07-21:
It's hard to guess what happened. Do you happen to have a packet dump of a few minutes that reproduces this issue?

eminence, 2020-07-21:
@nigelvh What satellite were you listening to?

nigelvh, 2020-07-21:
@pietern Unfortunately no. I have lots of storage and seems I tend to be around here enough. Recommendations on enabling a dump?

@eminence I am west coast. GOES-17.

matirko, 2022-04-07:
I am also having this issue on a Debian 11.3 VM, same byte counts as well when listening to GOES-16.

```
goesproc -c /usr/share/goestools/goesproc-goesr.conf -m packet  --subscribe tcp://127.0.0.1:5004
VC 8: M_SDU continuation failed; 34931 byte(s) needed to complete M_SDU, 188 byte(s) available
VC 8: M_SDU continuation failed; 34931 byte(s) needed to complete M_SDU, 188 byte(s) available
VC 8: M_SDU continuation failed; 34931 byte(s) needed to complete M_SDU, 188 byte(s) available
```

Vit < 150, drops are still nonzero though
```
2022-04-07T01:24:37Z [monitor] gain: 10.67, freq:  9023.7, omega: 2.159, vit(avg):  116, rs(sum):  570, packets: 48, drops:  3
2022-04-07T01:24:38Z [monitor] gain: 10.63, freq:  8976.3, omega: 2.158, vit(avg):   92, rs(sum):  232, packets: 50, drops:  4
2022-04-07T01:24:39Z [monitor] gain: 10.61, freq:  8994.7, omega: 2.158, vit(avg):  119, rs(sum):  509, packets: 41, drops:  7
...
```

For me, this issue cannot be fixed by restarting goespro - I still see the same error. 
...any ideas?

Here is the full log of the CMake build as well

```
root@debian:/home/mike# git clone https://github.com/pietern/goestools.git
cd goestools
git submodule init
git submodule update --recursive
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr ..

sudo make -j2 install 

Cloning into 'goestools'...
remote: Enumerating objects: 3056, done.
remote: Counting objects: 100% (156/156), done.
remote: Compressing objects: 100% (123/123), done.
remote: Total 3056 (delta 61), reused 83 (delta 29), pack-reused 2900
Receiving objects: 100% (3056/3056), 11.44 MiB | 4.15 MiB/s, done.
Resolving deltas: 100% (2020/2020), done.
Submodule 'vendor/libaec' (https://gitlab.dkrz.de/k202009/libaec.git) registered for path 'vendor/libaec'
Submodule 'vendor/libcorrect' (https://github.com/pietern/libcorrect.git) registered for path 'vendor/libcorrect'
Submodule 'vendor/nanomsg' (https://github.com/nanomsg/nanomsg.git) registered for path 'vendor/nanomsg'
Submodule 'vendor/sanitizers-cmake' (https://github.com/arsenm/sanitizers-cmake.git) registered for path 'vendor/sanitizers-cmake'
Submodule 'vendor/tinytoml' (https://github.com/mayah/tinytoml.git) registered for path 'vendor/tinytoml'
Cloning into '/home/mike/goestools/vendor/libaec'...
Cloning into '/home/mike/goestools/vendor/libcorrect'...
Cloning into '/home/mike/goestools/vendor/nanomsg'...
Cloning into '/home/mike/goestools/vendor/sanitizers-cmake'...
Cloning into '/home/mike/goestools/vendor/tinytoml'...
Submodule path 'vendor/libaec': checked out 'd808db1e1276f6c23feccdcefd56f7e38e2d7c9e'
Submodule path 'vendor/libcorrect': checked out 'a3c8890b43486822900a2c65a33701fe35bdc0e2'
Submodule path 'vendor/nanomsg': checked out '0c1aa2b288f6b167dbafe7e29c20e6fc7e71c000'
Submodule path 'vendor/sanitizers-cmake': checked out 'f09151bc87ba210f955d58dffc07b0ee1e755234'
Submodule path 'vendor/tinytoml': checked out '8fe890978b3390de1fc1f26c2d6bf1711c700c5a'
-- The CXX compiler identification is GNU 10.2.1
-- The C compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Defaulting build type to 'RelWithDebInfo'
-- Looking for dotprod in FEC
-- Looking for dotprod in FEC - not found
[... 399 lines cut ...]
[ 97%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/lrit_processor.cc.o
[ 97%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/options.cc.o
[ 98%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/packet_processor.cc.o
[ 98%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/string.cc.o
[ 98%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/proj.cc.o
[100%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/map_drawer.cc.o
[100%] Linking CXX executable goesproc
[100%] Built target goesproc
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/share/goestools/goesproc-goesn.conf
-- Installing: /usr/share/goestools/goesproc-goesr.conf
-- Up-to-date: /usr/share/goestools/wxstar
-- Installing: /usr/share/goestools/wxstar/wxstar_goes16_ch02_curve.png
-- Installing: /usr/share/goestools/wxstar/README.txt
-- Installing: /usr/share/goestools/wxstar/wxstar_goes16_lut.png
-- Up-to-date: /usr/share/goestools/ne
-- Installing: /usr/share/goestools/ne/ne_50m_admin_0_countries_lakes.json
-- Installing: /usr/share/goestools/ne/README.txt
-- Installing: /usr/share/goestools/ne/ne_50m_admin_1_states_provinces_lakes.json
-- Installing: /usr/bin/goesrecv
-- Installing: /usr/bin/goeslrit
-- Installing: /usr/bin/goesproc
-- Installing: /usr/bin/goespackets
```

matirko, 2022-04-07:
Also @pietern, was it just a wireshark dump on my --subscribe IP that you needed? Thanks!

pumatrax, 2022-04-08:
I just set up a new build with a rpi buster OS installed that I had for ham radio use. I get this error as well. Any suggestions????

pumatrax, 2022-04-08:
> Also @pietern, was it just a wireshark dump on my --subscribe IP that you needed? Thanks!

I am having the same issue. Any ideas?

pumatrax, 2022-04-08:
I get this during the compile of goesproc:

Scanning dependencies of target goesproc
[ 87%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/area.cc.o
In file included from /usr/include/c++/10/vector:72,
                 from /home/pi/goestools/src/goesrecv/publisher.h:5,
                 from /home/pi/goestools/src/goesrecv/packet_publisher.h:3,
                 from /home/pi/goestools/src/goesrecv/config.h:6,
                 from /home/pi/goestools/src/goesrecv/config.cc:1:
/usr/include/c++/10/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {toml::Value}; _Tp = toml::Value; _Alloc = std::allocator<toml::Value>]’:
/usr/include/c++/10/bits/vector.tcc:426:7: note: parameter passing for argument of type ‘std::vector<toml::Value>::iterator’ changed in GCC 7.1
  426 |       vector<_Tp, _Alloc>::
      |       ^~~~~~~~~~~~~~~~~~~
/usr/include/c++/10/bits/vector.tcc: In member function ‘toml::Value* toml::Value::push(toml::Value&&)’:
/usr/include/c++/10/bits/vector.tcc:121:21: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<toml::Value*, std::vector<toml::Value> >’ changed in GCC 7.1
  121 |    _M_realloc_insert(end(), std::forward<_Args>(__args)...);
      |    ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/10/map:60,
                 from /home/pi/goestools/vendor/tinytoml/include/toml/toml.h:18,
                 from /home/pi/goestools/src/goesrecv/config.cc:7:
/usr/include/c++/10/bits/stl_tree.h: In member function ‘std::pair<std::_Rb_tree_node_base*, std::_Rb_tree_node_base*> std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_get_insert_hint_unique_pos(std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::const_iterator, const key_type&) [with _Key = std::__cxx11::basic_string<char>; _Val = std::pair<const std::__cxx11::basic_string<char>, toml::Value>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, toml::Value> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, toml::Value> >]’:
/usr/include/c++/10/bits/stl_tree.h:2193:5: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, toml::Value>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, toml::Value> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, toml::Value> > >::const_iterator’ changed in GCC 7.1
 2193 |     _Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/10/bits/stl_tree.h: In function ‘std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::iterator std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_emplace_hint_unique(std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::const_iterator, _Args&& ...) [with _Args = {const std::piecewise_construct_t&, std::tuple<const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&>, std::tuple<>}; _Key = std::__cxx11::basic_string<char>; _Val = std::pair<const std::__cxx11::basic_string<char>, toml::Value>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, toml::Value> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, toml::Value> >]’:
/usr/include/c++/10/bits/stl_tree.h:2458:7: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, toml::Value>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, toml::Value> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, toml::Value> > >::const_iterator’ changed in GCC 7.1
 2458 |       _Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/10/bits/stl_tree.h:2458:7: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, toml::Value>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, toml::Value> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, toml::Value> > >::const_iterator’ changed in GCC 7.1
In file included from /usr/include/c++/10/map:61,
                 from /home/pi/goestools/vendor/tinytoml/include/toml/toml.h:18,
                 from /home/pi/goestools/src/goesrecv/config.cc:7:
/usr/include/c++/10/bits/stl_map.h: In member function ‘toml::Value* toml::Value::setChild(const string&, toml::Value&&)’:
/usr/include/c++/10/bits/stl_map.h:501:37: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, toml::Value>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, toml::Value> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, toml::Value> > >::const_iterator’ changed in GCC 7.1
  501 |    __i = _M_t._M_emplace_hint_unique(__i, std::piecewise_construct,
[... cut ...]
/usr/include/c++/10/bits/vector.tcc:121:21: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<nlohmann::basic_json<>*, std::vector<nlohmann::basic_json<>, std::allocator<nlohmann::basic_json<> > > >’ changed in GCC 7.1
  121 |    _M_realloc_insert(end(), std::forward<_Args>(__args)...);
      |    ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 90%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/datagram_socket.cc.o
[ 90%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/filename.cc.o
[ 91%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/source.cc.o
[ 91%] Linking CXX executable goesrecv
[ 91%] Built target goesrecv

pietern, 2022-04-15:
@matirko I suspect the `continuation failed` messages are caused by the drops. A single one of these packets is built from multiple packets, so if one or more in a sequence get dropped, you're bound to get errors like this.

If this happens on a stream without packet drops then more investigation will be needed.

pumatrax, 2022-04-15:
@pietern Thats the odd part about it, there isnt drops. maybe errors in the data stream but no drops. 

I have seen the continuation failf disappear at any point. In a waterfall test I see the carrier on 1694.1 and its strong but I dont see much of the wide hump and a strong single in the data stream.

2022-04-15T23:09:40Z [monitor] gain:  4.69, freq: -5545.3, omega: 2.593, vit(avg):  482, rs(sum):  597, packets: 56, drops:  0
2022-04-15T23:09:41Z [monitor] gain:  4.69, freq: -5531.1, omega: 2.593, vit(avg):  483, rs(sum):  631, packets: 57, drops:  0
2022-04-15T23:09:42Z [monitor] gain:  4.70, freq: -5545.7, omega: 2.593, vit(avg):  487, rs(sum):  662, packets: 57, drops:  0
2022-04-15T23:09:43Z [monitor] gain:  4.70, freq: -5550.8, omega: 2.593, vit(avg):  487, rs(sum):  591, packets: 56, drops:  0
2022-04-15T23:09:44Z [monitor] gain:  4.70, freq: -5551.7, omega: 2.594, vit(avg):  487, rs(sum):  797, packets: 57, drops:  0
2022-04-15T23:09:45Z [monitor] gain:  4.70, freq: -5543.7, omega: 2.593, vit(avg):  481, rs(sum):  558, packets: 57, drops:  0
2022-04-15T23:09:46Z [monitor] gain:  4.70, freq: -5551.1, omega: 2.593, vit(avg):  478, rs(sum):  631, packets: 55, drops:  0
2022-04-15T23:09:47Z [monitor] gain:  4.69, freq: -5575.3, omega: 2.591, vit(avg):  491, rs(sum):  692, packets: 57, drops:  0
2022-04-15T23:09:48Z [monitor] gain:  4.70, freq: -5536.3, omega: 2.587, vit(avg):  493, rs(sum):  736, packets: 58, drops:  0
2022-04-15T23:09:49Z [monitor] gain:  4.70, freq: -5545.6, omega: 2.586, vit(avg):  493, rs(sum):  675, packets: 55, drops:  0
2022-04-15T23:09:50Z [monitor] gain:  4.70, freq: -5549.8, omega: 2.587, vit(avg):  494, rs(sum):  759, packets: 57, drops:  0
2022-04-15T23:09:51Z [monitor] gain:  4.71, freq: -5540.4, omega: 2.585, vit(avg):  489, rs(sum):  726, packets: 58, drops:  0
2022-04-15T23:09:52Z [monitor] gain:  4.70, freq: -5554.9, omega: 2.585, vit(avg):  495, rs(sum):  729, packets: 55, drops:  0
2022-04-15T23:09:53Z [monitor] gain:  4.70, freq: -5558.5, omega: 2.586, vit(avg):  488, rs(sum):  644, packets: 56, drops:  0
2022-04-15T23:09:54Z [monitor] gain:  4.69, freq: -5540.0, omega: 2.587, vit(avg):  482, rs(sum):  702, packets: 59, drops:  0
2022-04-15T23:09:55Z [monitor] gain:  4.70, freq: -5562.4, omega: 2.586, vit(avg):  489, rs(sum):  635, packets: 56, drops:  0
2022-04-15T23:09:56Z [monitor] gain:  4.70, freq: -5563.1, omega: 2.585, vit(avg):  492, rs(sum):  665, packets: 55, drops:  0
2022-04-15T23:09:57Z [monitor] gain:  4.70, freq: -5572.9, omega: 2.585, vit(avg):  490, rs(sum):  705, packets: 59, drops:  0
2022-04-15T23:09:58Z [monitor] gain:  4.70, freq: -5593.0, omega: 2.585, vit(avg):  497, rs(sum):  734, packets: 56, drops:  0
2022-04-15T23:09:59Z [monitor] gain:  4.70, freq: -5577.8, omega: 2.585, vit(avg):  489, rs(sum):  608, packets: 55, drops:  0
2022-04-15T23:10:00Z [monitor] gain:  4.70, freq: -5575.5, omega: 2.587, vit(avg):  487, rs(sum):  655, packets: 56, drops:  0
2022-04-15T23:10:01Z [monitor] gain:  4.70, freq: -5578.5, omega: 2.587, vit(avg):  485, rs(sum):  681, packets: 59, drops:  0
2022-04-15T23:10:02Z [monitor] gain:  4.71, freq: -5569.9, omega: 2.587, vit(avg):  470, rs(sum):  544, packets: 56, drops:  0

pietern, 2022-04-16:
Thanks for confirming. I don't see this on the GOES-17 stream that @creinemann gave me access to. It is possible that this is a ground station problem and that you can safely ignore the warning. Can you receive the products on other VCIDs?

Either way, I'd be happy to take a peek and figure out what's happening by looking at the packet stream.

You can capture packets with:

```
goespackets --subscribe [ADDRESS] --record
```

After a few minutes you can press Ctrl-C to interrupt and stop the capture.

Please make sure you see the `continuation failed` message in the goesproc output while the capture is happening.

pietern, 2022-04-16:
FYI, for GOES-17 this is the VCID layout:

```
The GOES-17 HRIT/EMWIN West broadcast utilizes virtual channels for the transmission of
several simultaneous data streams. The table below identifies the current
virtual channels assignments.

    HRIT/EMWIN
Virtual Channel ID     Group          Product Name
------------------    -------    -------------------------------
        0              Imagery     Admin Text Messages
        1             *Imagery     Mesoscale 0.5 � 2km (ch. 2, 7 and 13)
        2             *Imagery     Band 2 - Visual
        7             *Imagery     Band 7 - Shortwave Infrared
        8             *Imagery     Band 8 � Mid-Level Tropospheric Water Vapor
        9             *Imagery     Band 9 � Upper-level Tropospheric Water Vapor
       13             *Imagery     Band 13 � Clean Longwave Infrared
       14             *Imagery     Band 14 � Infrared Longwave
       15             *Imagery     Band 15 � Dirty Longwave Infrared
       16             *Imagery     Band 13 � GOES-16 4km Clean Longwave Infrared
       20              EMWIN       High Priority
       21              EMWIN       Graphics
       22              EMWIN       Low Priority
       24              Imagery     NHC Graphics Products
       25             *Imagery     Other GOES-17 Level II Products
       30              DCS         DCS Admin
       32              DCS         DCS Data
       60              Imagery     Himawari-8 4km Infrared, Water Vapor and Visual

*NOTE* - Data Quality Flags are not made available for any of the GOES Imagery products.

For additional broadcast information or questions please visit https://www.noaasis.noaa.gov/ or contact the
HRIT/EMWIN Broadcast Manager at seth.clevenstine@noaa.gov.
```

pietern, 2022-04-16:
@pumatrax @matirko Can you both confirm the version of goestools you're on (output of `goesproc --version`)?

Also, does the `continuation failed` message show up every now and then or do you see a continuous stream?

The former could indicate a ground station problem and other products should arrive just fine. The latter could indicate a problem in goestools itself.

pumatrax, 2022-04-16:
@pietern
[goespackets.zip](https://github.com/pietern/goestools/files/8500333/goespackets.zip)
[image]

Hi, here is a packet capture. I dont know what tools to use to inspect it.  The continuation failed error is consistent no matter how good the signal is. last night my vit was around 250-300.  Hope this helps.

pumatrax, 2022-04-16:
@pietern 
> `goesproc --version`)?

goesproc -- 9f21e52 (Sun, 27 Mar 2022 14:00:22 +0200)

pietern, 2022-04-16:
@pumatrax The packets you shared are all identical. That's why the error keeps repeating. I have no idea why goesrecv is producing packets for what appears to be a bad signal. I also see the "phase flip detected" message in your goesrecv output. This never happens on a good signal. Another indicator of a bad signal is that the number of Reed-Solomon corrections is very high (>10 per packet) for the signal quality you would expect from ~400 Viterbi corrections.

Can you share a screenshot of the waterfall?

pietern, 2022-04-16:
Can you also post your copy of `goesrecv.conf` and share hardware details of your setup?

nigelvh, 2022-04-16:
@pumatrax For reference here's example output from my setup to describe what @pietern is talking about. I have a much lower vit than you do, and the rs corrections are very few.

```
2022-04-16T18:07:07Z [monitor] gain: 14.79, freq:  7875.8, omega: 2.589, vit(avg):  111, rs(sum):    0, packets: 55, drops:  0
2022-04-16T18:07:08Z [monitor] gain: 14.81, freq:  7883.6, omega: 2.589, vit(avg):  110, rs(sum):    0, packets: 58, drops:  0
2022-04-16T18:07:09Z [monitor] gain: 14.83, freq:  7896.4, omega: 2.589, vit(avg):  108, rs(sum):    0, packets: 57, drops:  0
2022-04-16T18:07:10Z [monitor] gain: 14.82, freq:  7873.4, omega: 2.589, vit(avg):  109, rs(sum):    2, packets: 55, drops:  0
2022-04-16T18:07:11Z [monitor] gain: 14.82, freq:  7879.8, omega: 2.589, vit(avg):  117, rs(sum):    3, packets: 58, drops:  0
2022-04-16T18:07:12Z [monitor] gain: 14.80, freq:  7890.6, omega: 2.589, vit(avg):  119, rs(sum):    3, packets: 56, drops:  0
2022-04-16T18:07:13Z [monitor] gain: 14.84, freq:  7872.1, omega: 2.589, vit(avg):  111, rs(sum):    1, packets: 56, drops:  0
2022-04-16T18:07:14Z [monitor] gain: 14.87, freq:  7885.2, omega: 2.589, vit(avg):  117, rs(sum):    0, packets: 57, drops:  0
2022-04-16T18:07:15Z [monitor] gain: 14.88, freq:  7862.8, omega: 2.589, vit(avg):  114, rs(sum):    0, packets: 57, drops:  0
2022-04-16T18:07:16Z [monitor] gain: 14.85, freq:  7864.8, omega: 2.589, vit(avg):  114, rs(sum):    0, packets: 56, drops:  0
2022-04-16T18:07:17Z [monitor] gain: 14.82, freq:  7880.3, omega: 2.589, vit(avg):  115, rs(sum):    0, packets: 57, drops:  0
```

nigelvh, 2022-04-16:
I can't speak to why the same packet appears to be repeating, but I would suggest further work to improve signal into your setup aside from that.

pietern, 2022-04-16:
@nigelvh Thanks for posting what a good signal looks like :)

Did you ever experience this problem again since you originally posted the issue, by the way?

pumatrax, 2022-04-16:
@pietern 

Okay here is the run down:

I am using a the 1.9GHz parabolic antenna. 
Noolec Saw filter and lna
Rtl-sdr dongle. 
Receiving GOES16
Location: New York City area

Hosts:
Started with RPi 4 with debian releases and even tried ubuntu on it.
Currently running a Virtualbox VM on Windows 10 laptop to rule out hosts, OS's, etc....

The phase flip seems to be happening more on this VM then the RPi, also sun is high in the sky now. I was getting 250 vit last night with with rs corrections as low as 50-90. I see with @nigelvh its still not good enough. 

I mentioned  in earlier post that the waterfall does not look right. I had this set up last year in another location and waterfall looked great. The carrier signal at 1694.1 is strong but the wide 1.205 MHz hump is not really present. 
[image]

The antenna is on a tripod on my balcony. Its the only place I cant put it. Maybe I could take the setup to a mall parking lot and try there in the clear. 
[image]

goesrecv.conf

`[demodulator]
mode = "hrit"
source = "rtlsdr"

[rtlsdr]
frequency = 1694100000
sample_rate = 2000000
gain = 50
bias_tee = true

[costas]
max_deviation = 200e3

[decoder.packet_publisher]
bind = "tcp://0.0.0.0:5004"
send_buffer = 1048576

[monitor]
statsd_address = "udp4://localhost:8125"

[clock_recovery.sample_publisher]
bind = "tcp://0.0.0.0:5002"
send_buffer = 2097152

[demodulator.stats_publisher]
bind = "tcp://0.0.0.0:6001"

[decoder.stats_publisher]
bind = "tcp://0.0.0.0:6002"`

pietern, 2022-04-16:
Could you zoom in vertically on that waterfall? It's hard to make out the size in dB of the signal around 1694.1 MHz.

Everything else looks nominal to me.

What I don't understand is how this can happen if you still see lower Viterbi and Reed-Solomon corrections at night...

pietern, 2022-04-16:
You could try using a lower gain value. It is possible you're overloading the LNA in the SDR itself.

Some SDRs are better than others (e.g. the E4000's are notorious for not always doing great at higher frequencies). Which tuner chip is in the SDR you're using?

pumatrax, 2022-04-16:
I might be pointing at the wrong satellite. I always felt I was too south with my direction that gave me the lowest vit but based on the directionfinder app I needed to be more SE. I move it to that position again and the vit dropped below 2000 and now I see a little of the hump in the waterfall. 

I guess if I cant get the vit low enough from the balcony theres not much I can do. but I think I solved the problem by realizing I was not in the right location. I also wasnt too SW to be receiving GOES17. Maybe I was getting carrier of GOES 14 where it is straight south from my location.  

[image]

pietern, 2022-04-16:
I was just about to say the same. GOES-18 is being checked out at 89.5 degrees west and I remember seeing a carrier transmitted on GOES-17 when it was being checked out. The earlier waterfall you shared did have that big carrier and that might just produce valid packets...

pietern, 2022-04-16:
GOES-17 is at 75.2 degrees west, FWIW. If you're in the NYC area that means you might need to point further east.

pumatrax, 2022-04-16:
I am trying to receive GOES-16. 17 it too low of an elevation for me, I dont have a clear line of sight.  I am seeing a normal signal now, its just too weak. 1600 VIT, no good. I can try to amplify it without overloading but who knows. 

My house has solar panels on it and being on the balcony there might be too much cover and its getting in the way of signal.

pietern, 2022-04-16:
Right, I wrote GOES-17 at 75.2 but meant GOES-16...

In the waterfall you should see about 8dB over the noise floor for it to work. You could try playing with the gain value in `goesrecv.conf` a bit. Sometimes even lowering helps improving the signal quality (even though it's quieter).

pumatrax, 2022-04-16:
eureka! I found it. I pushed the tripod forward so it can be further out from the balcony and started adjusting. still a poor signal but no continuation failed error because I am pointing the right place. I learned about GOES-14 last night and how its in storage now so that strong carrier with no data stream must be from it. 

[image]

creinemann, 2022-04-16:
A few days late, but--Just a couple of comments, From your setup location pic, I would agree that getting away from your patio would be a huge help. Maybe a handrailing clamp-on tv mount to get the dish away from all that metal. (looks like metal siding all around, the posts, roof, etc.  And solar panels and more importantly the inverters can be very problematic for RF interference.  I gave a presentation to NOAA just last week, and RF Interference is becoming quite the issue with the ASW-3 band for 4G and 5G rollout happening all over. I have a post about some of it on my site at: [https://usradioguy.com/satellites/aws-3-interference-and-hrit-grb-hrpt/](https://usradioguy.com/satellites/aws-3-interference-and-hrit-grb-hrpt/  )

pietern, 2022-04-16:
Congrats!

The carrier is more likely to be GOES-18. It was launched last month and is undergoing checkout and testing before it will drift to 137 degrees west where it will eventually replace GOES-17.

The N-series sats (13-15) used the LRIT signal on 1691 MHz. The R-series sats (16-19) use the HRIT signal on 1694.1 MHz.

nigelvh, 2022-04-16:
@pumatrax I will second what @pietern said, different SDRs with ostensibly the same chip can have vastly different performance. I tried a few before I settled on one that worked significantly better than the others. Otherwise I took a 2.4GHz dish, modified the feed on it, and strapped some extra grid sections on it to make it larger and capture more signal. I SAWbird+ GOES LNA+Preamp. I have photos and more detail on my setup here: https://digitalnigel.com/wordpress/?p=2932

@pietern Thanks for checking in, no, it seems like the issue hasn't recurred. Probably something temporary on the downstream from the sat.

nigelvh, 2022-04-16:
Also, pointing is extremely critical. The compass will get you in the ball park, but I've found that hair breadth adjustments will make significant differences, so take some time to tweak your aim in both the horizontal and vertical for best numbers.

pumatrax, 2022-04-16:
@pietern @creinemann @nigelvh 

thank you all for chiming in and helping. What a relief after a couple of days thinking it may be software related and trying different flavors of hardware and software.  Thinking I was pointing at the right satellite had me tweaking and doing everything to no end and even got down to 200 VIT. But I guess it was a good learning experience. I did see big differences between dongles (I have quite a few) and even cable hook ups.  It would be great if there was support for SDRplay Devices because signal to noise ratio and signal strength  was so amazing when I had it hooked up and pointing at that strong carrier.  

I am definitely going to do a clamp mount to put on the fence, I have some gaffing equipment laying around. Film production equipment comes in handy for doing quick set ups with tripods and mounting things to things. 

I know some others are having this issue so it’s good to know that they’ll just have to keep pointing to find the right satellite. 😎

pietern, 2022-04-16:
Great to hear, @pumatrax. Good luck with the rest of the experimentation!

Given that we now know what was causing the original issue, I'm closing the issue.

closed 2022-04-16 by pietern