pietern/goestools #92: LRIT Runtime Error in goesproc

nigelvh, 2020-07-21:
Got this error just a few minutes ago (just around 9AM Pacific).
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Assertion `it != m.end()` failed at /home/nigel/goestools/src/lrit/lrit.h:165
Aborted (core dumped)
```
I'm on GOES17.

Currently running version 543ece3dc99ac865d87ac03490be99cf1f0457a7 from the testing in https://github.com/pietern/goestools/issues/77

Timing of this issue is close to https://github.com/pietern/goestools/issues/91, but I don't have anything else to assume they're related.

nigelvh, 2020-07-21:
Another instance here around 13:00 Pacific. Seems like nearly (if not exactly), 4hrs apart. So perhaps some new scheduled data is coming off the satellite that's not being handled well.

loudelatorre, 2020-07-21:
I just had that issue:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Assertion `it != m.end()` failed at /home/pi/goestools/src/lrit/lrit.h:165
/home/pi/startgoesproc.sh: line 2:   762 Aborted                 /usr/bin/goesproc -c /usr/share/goestools/goesproc-goesr.conf -m packet --subscribe tcp://127.0.0.1:5004 --out /home/pi/goes
```