pietern/goestools #95: CV_LOAD_IMAGE_UNCHANGED was not declared in this scope

kidshare, 2020-08-31:
What do I do with this error. I have not seen it before. I'm moving from a desktop to a Pi4, this build failed.

/home/pi4/Desktop/goestools/src/goesproc/config.cc:210:37: error: ‘CV_LOAD_IMAGE_UNCHANGED’ was not declared in this scope
  210 |         auto img = cv::imread(path, CV_LOAD_IMAGE_UNCHANGED);
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:76: src/goesproc/CMakeFiles/goesproc.dir/config.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2360: src/goesproc/CMakeFiles/goesproc.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

pietern, 2020-09-01:
Which OS distribution are you running? Which version of OpenCV is the build trying to use?

kidshare, 2020-09-01:
This is an install of Ubuntu Mate 20.0.4. After initial install I used apt-get install build-essential cmake libusb-1.0 libproj-dev and libopencv-dev. It installed OpenCV version 4 and Proj version 6.

The OpenCV4 gave me the error referred to in issue #90. Making the change listed there eliminated that error but then this one came up. After hunting around similar errors in Google I made this change to /src/goesproc/config.cc in line 210

auto img = cv::imread(path, CV_LOAD_IMAGE_UNCHANGED);
is now
auto img = cv::imread(path, cv::IMREAD_UNCHANGED);

just changing it to IMREAD_UNCHANGED didn't work, had to add the cv::

This compiled OK but then other issues came up involving proj version 6.3.1. Those are addressed in issue #83. I used this line in /goesproc/proj.h, changing the 5 to a 6:

#if PROJ_VERSION_MAJOR < 4 || PROJ_VERSION_MAJOR > 6

That worked, apparently version 6.3 works fine. Now the errors were due to deprecated proj, but added this line also referred to in #83 to handler_goesr.cc, handler_goesn.cc, handler_himawari8.cc, proj.cc and proj.h

#define ACCEPT_USE_OF_DEPRECATED_PROJ_API_H 1

That cleared everything up. I copied over the conf files from the desktop computer (which I had set up on Ubuntu 18 in Nov. 2018) and some other minor issues came up in the goesrecv.conf file:

maxdeviation=200e3 had to be changed to max_deviation=200e3
[decoder.packet-publisher] had to be changed to [decoder.packet_publisher]
bind="tcp://0.0.0.5004" had to be changed to bind="tcp://0.0.0.0:5004"

Finally, the last changes were to modify goesproc-goesr.conf to store the images on an external laptop drive so I don't tear up the SD card. Wherever folder storage is mentioned in that file I added /media/username/externaldrive to the beginning. Now everything is working fine and I'm saving some electricity.

Thanks!

closed 2020-09-01 by kidshare

pietern, 2022-02-13:
Thanks for the pointers for fixing compilation with newer dependencies.

I'm making a pass over the PRs and issues and this is super helpful!