pietern/goestools #96: Multiple errors while compiling goesproc (temporary solution)

MaxPastushkov, 2020-09-10:
I wanted to compile goestools on Arch Linux, and I ran into a series of problems. I'm not sure if anyone else is having this problem, but here goes anyway.

First of all, the packages I needed were proj, rtl-sdr, airspy, and opencv. There's probably some more, but these are the main ones. Also, make sure to clone the repo with the `--recurse-submodules` parameter. Next, when I tried compiling goesproc, I got the following error:

> /home/max/Downloads/goestools/src/lib/packet_reader.h:11:38: error: ‘uint8_t’ was not declared in this scope
   11 |   virtual bool nextPacket(std::array<uint8_t, 892>& out) = 0;
      |                                                     ^~~~~~~
/home/max/Downloads/goestools/src/lib/packet_reader.h:11:50: error: template argument 1 is invalid
   11 |   virtual bool nextPacket(std::array<uint8_t, 892>& out) = 0;

After some screwing around, I managed to fix it, but there were still more problems. Goestools was written for OpenCV 3.2, while the newest version (and the one in the Arch repos) is 4.4. In the end, I managed to fix the problem (although kind-of a bodge). Here are the steps:

1. Comment out (#) or remove lines 41 and 43 of goestools/src/goesproc/CMakeLists.txt (if and endif statements).
2. Add the line `#include <stdint.h>` to goestools/src/lib/packet_reader.h next to the other include(s).
3. Add the line `#define ACCEPT_USE_OF_DEPRECATED_PROJ_API_H` in goestools/src/goesproc/proj.h on line 6 (or anywhere **before** the includes).
4. Replace `CV_LOAD_IMAGE_UNCHANGED` in goestools/src/goesproc/config.cc on line 210 (starting with `auto img =`) with `cv::IMREAD_UNCHANGED`.

Keep in mind that the line numbers might change, so if something seems wrong just look at the description I wrote for each line. Also, excuse me for the crude way of creating a patch; I wasn't able to make a proper .patch file. In any case, I hope this really helped someone. Feel free to ask questions!

cottsay, 2020-10-14:
Please take a look at these PRs to see if they resolve your issues:
2. #81

pietern, 2022-03-15:
Thanks for submitting the changes needed, @MaxPastushkov. All modifications are now in the main branch.

closed 2022-03-15 by pietern