pietern/goestools #97: drops

hemna, 2020-09-16:
Hey guys,
 I've had goestools running for a long while on my rpi in a docker container.   I've been noticing that I've not been able to decode any packets for a while due to drops.   I'm seeing vit rates between 280 and 500, all with 13+ or so drops.
I'm not sure what's up.  Am I just not getting good enough signal to decode anything?   

```
2020-09-16T21:02:42Z [monitor] gain: 11.80, freq:  5458.6, omega: 2.589, vit(avg):  377, rs(sum):    0, packets: 19, drops: 12
2020-09-16T21:02:43Z [monitor] gain: 11.80, freq:  5449.3, omega: 2.589, vit(avg):  502, rs(sum):    0, packets: 18, drops: 13
2020-09-16T21:02:44Z [monitor] gain: 11.80, freq:  5445.1, omega: 2.589, vit(avg):  335, rs(sum):    2, packets: 18, drops: 13
2020-09-16T21:02:45Z [monitor] gain: 11.80, freq:  5449.5, omega: 2.589, vit(avg):  267, rs(sum):    9, packets: 19, drops: 12
2020-09-16T21:02:46Z [monitor] gain: 11.79, freq:  5433.0, omega: 2.589, vit(avg):  362, rs(sum):    9, packets: 19, drops: 13
2020-09-16T21:02:47Z [monitor] gain: 11.80, freq:  5452.4, omega: 2.589, vit(avg):  489, rs(sum):    0, packets: 17, drops: 13
2020-09-16T21:02:48Z [monitor] gain: 11.80, freq:  5462.3, omega: 2.589, vit(avg):  409, rs(sum):    0, packets: 19, drops: 13
2020-09-16T21:02:49Z [monitor] gain: 11.79, freq:  5446.1, omega: 2.589, vit(avg):  370, rs(sum):    4, packets: 19, drops: 12
2020-09-16T21:02:50Z [monitor] gain: 11.78, freq:  5440.6, omega: 2.589, vit(avg):  553, rs(sum):    0, packets: 18, drops: 13
2020-09-16T21:02:51Z [monitor] gain: 11.77, freq:  5455.1, omega: 2.589, vit(avg):  451, rs(sum):    3, packets: 19, drops: 13
2020-09-16T21:02:52Z [monitor] gain: 11.77, freq:  5458.4, omega: 2.589, vit(avg):  390, rs(sum):    1, packets: 20, drops: 12
2020-09-16T21:02:53Z [monitor] gain: 11.78, freq:  5436.7, omega: 2.589, vit(avg):  456, rs(sum):    0, packets: 17, drops: 13
2020-09-16T21:02:54Z [monitor] gain: 11.80, freq:  5445.8, omega: 2.589, vit(avg):  381, rs(sum):    2, packets: 19, drops: 12
2020-09-16T21:02:55Z [monitor] gain: 11.80, freq:  5469.4, omega: 2.589, vit(avg):  574, rs(sum):    4, packets: 18, drops: 14
2020-09-16T21:02:56Z [monitor] gain: 11.80, freq:  5444.6, omega: 2.589, vit(avg):  383, rs(sum):    3, packets: 18, drops: 13
2020-09-16T21:02:57Z [monitor] gain: 11.80, freq:  5451.7, omega: 2.589, vit(avg):  449, rs(sum):    4, packets: 19, drops: 12
2020-09-16T21:02:58Z [monitor] gain: 11.80, freq:  5460.8, omega: 2.589, vit(avg):  562, rs(sum):    2, packets: 19, drops: 13
2020-09-16T21:02:59Z [monitor] gain: 11.80, freq:  5464.0, omega: 2.589, vit(avg):  394, rs(sum):    3, packets: 17, drops: 13
2020-09-16T21:03:00Z [monitor] gain: 11.80, freq:  5462.3, omega: 2.589, vit(avg):  433, rs(sum):    2, packets: 19, drops: 12
2020-09-16T21:03:01Z [monitor] gain: 11.79, freq:  5470.9, omega: 2.589, vit(avg):  444, rs(sum):    0, packets: 19, drops: 13
2020-09-16T21:03:02Z [monitor] gain: 11.78, freq:  5451.0, omega: 2.589, vit(avg):  426, rs(sum):    1, packets: 18, drops: 13
2020-09-16T21:03:03Z [monitor] gain: 11.80, freq:  5455.3, omega: 2.589, vit(avg):  445, rs(sum):    3, packets: 17, drops: 12
2020-09-16T21:03:04Z [monitor] gain: 11.80, freq:  5464.5, omega: 2.589, vit(avg):  328, rs(sum):    0, packets: 22, drops: 12
2020-09-16T21:03:05Z [monitor] gain: 11.80, freq:  5467.4, omega: 2.589, vit(avg):  354, rs(sum):    0, packets: 19, drops: 13
2020-09-16T21:03:06Z [monitor] gain: 11.80, freq:  5462.6, omega: 2.589, vit(avg):  274, rs(sum):    6, packets: 20, drops: 12
2020-09-16T21:03:07Z [monitor] gain: 11.80, freq:  5445.3, omega: 2.589, vit(avg):  526, rs(sum):    0, packets: 19, drops: 13
2020-09-16T21:03:08Z [monitor] gain: 11.81, freq:  5464.5, omega: 2.589, vit(avg):  363, rs(sum):    6, packets: 18, drops: 13
2020-09-16T21:03:09Z [monitor] gain: 11.80, freq:  5485.2, omega: 2.589, vit(avg):  476, rs(sum):    3, packets: 19, drops: 12
2020-09-16T21:03:10Z [monitor] gain: 11.80, freq:  5421.3, omega: 2.589, vit(avg):  538, rs(sum):    4, packets: 19, drops: 13
2020-09-16T21:03:11Z [monitor] gain: 11.79, freq:  5451.4, omega: 2.589, vit(avg):  720, rs(sum):    0, packets: 19, drops: 13
2020-09-16T21:03:12Z [monitor] gain: 11.79, freq:  5463.6, omega: 2.589, vit(avg):  515, rs(sum):    5, packets: 18, drops: 13
2020-09-16T21:03:13Z [monitor] gain: 11.79, freq:  5447.1, omega: 2.589, vit(avg):  444, rs(sum):    3, packets: 19, drops: 12
2020-09-16T21:03:14Z [monitor] gain: 11.78, freq:  5436.9, omega: 2.589, vit(avg):  453, rs(sum):    2, packets: 18, drops: 13
2020-09-16T21:03:15Z [monitor] gain: 11.77, freq:  5432.7, omega: 2.589, vit(avg):  399, rs(sum):    3, packets: 18, drops: 13
2020-09-16T21:03:16Z [monitor] gain: 11.77, freq:  5461.3, omega: 2.589, vit(avg):  378, rs(sum):    2, packets: 18, drops: 13
2020-09-16T21:03:17Z [monitor] gain: 11.76, freq:  5456.4, omega: 2.589, vit(avg):  393, rs(sum):   10, packets: 19, drops: 12
2020-09-16T21:03:18Z [monitor] gain: 11.75, freq:  5458.8, omega: 2.589, vit(avg):  382, rs(sum):    0, packets: 19, drops: 13
2020-09-16T21:03:19Z [monitor] gain: 11.75, freq:  5454.7, omega: 2.589, vit(avg):  521, rs(sum):    6, packets: 17, drops: 13
2020-09-16T21:03:20Z [monitor] gain: 11.74, freq:  5452.2, omega: 2.589, vit(avg):  416, rs(sum):    2, packets: 20, drops: 12
2020-09-16T21:03:21Z [monitor] gain: 11.74, freq:  5463.2, omega: 2.589, vit(avg):  553, rs(sum):    0, packets: 18, drops: 13
2020-09-16T21:03:22Z [monitor] gain: 11.76, freq:  5448.6, omega: 2.589, vit(avg):  432, rs(sum):    1, packets: 18, drops: 13
2020-09-16T21:03:23Z [monitor] gain: 11.75, freq:  5478.3, omega: 2.589, vit(avg):  341, rs(sum):    6, packets: 18, drops: 13
2020-09-16T21:03:24Z [monitor] gain: 11.75, freq:  5453.3, omega: 2.589, vit(avg):  299, rs(sum):   10, packets: 19, drops: 13
2020-09-16T21:03:25Z [monitor] gain: 11.75, freq:  5452.5, omega: 2.589, vit(avg):  444, rs(sum):    0, packets: 18, drops: 13
2020-09-16T21:03:26Z [monitor] gain: 11.76, freq:  5444.3, omega: 2.589, vit(avg):  481, rs(sum):    0, packets: 18, drops: 13
2020-09-16T21:03:27Z [monitor] gain: 11.77, freq:  5456.2, omega: 2.589, vit(avg):  431, rs(sum):    0, packets: 19, drops: 12
2020-09-16T21:03:28Z [monitor] gain: 11.77, freq:  5463.2, omega: 2.589, vit(avg):  489, rs(sum):    2, packets: 18, drops: 13
2020-09-16T21:03:29Z [monitor] gain: 11.76, freq:  5453.2, omega: 2.589, vit(avg):  307, rs(sum):    3, packets: 18, drops: 13
2020-09-16T21:03:30Z [monitor] gain: 11.78, freq:  5438.0, omega: 2.589, vit(avg):  285, rs(sum):    3, packets: 19, drops: 12
2020-09-16T21:03:31Z [monitor] gain: 11.77, freq:  5460.0, omega: 2.589, vit(avg):  417, rs(sum):    5, packets: 19, drops: 13
2020-09-16T21:03:32Z [monitor] gain: 11.76, freq:  5464.0, omega: 2.589, vit(avg):  445, rs(sum):    3, packets: 18, drops: 13
2020-09-16T21:03:33Z [monitor] gain: 11.76, freq:  5469.0, omega: 2.589, vit(avg):  248, rs(sum):    3, packets: 18, drops: 13
2020-09-16T21:03:34Z [monitor] gain: 11.76, freq:  5456.9, omega: 2.589, vit(avg):  312, rs(sum):    8, packets: 19, drops: 12
2020-09-16T21:03:35Z [monitor] gain: 11.76, freq:  5461.4, omega: 2.589, vit(avg):  303, rs(sum):    0, packets: 18, drops: 13
2020-09-16T21:03:36Z [monitor] gain: 11.76, freq:  5441.6, omega: 2.589, vit(avg):  434, rs(sum):    1, packets: 18, drops: 13
2020-09-16T21:03:37Z [monitor] gain: 11.75, freq:  5459.9, omega: 2.589, vit(avg):  325, rs(sum):    1, packets: 18, drops: 13
2020-09-16T21:03:38Z [monitor] gain: 11.75, freq:  5458.6, omega: 2.589, vit(avg):  365, rs(sum):   10, packets: 19, drops: 13
2020-09-16T21:03:39Z [monitor] gain: 11.75, freq:  5451.9, omega: 2.589, vit(avg):  409, rs(sum):    0, packets: 17, drops: 13
2020-09-16T21:03:40Z [monitor] gain: 11.75, freq:  5472.9, omega: 2.589, vit(avg):  448, rs(sum):    0, packets: 18, drops: 12
2020-09-16T21:03:41Z [monitor] gain: 11.76, freq:  5454.3, omega: 2.589, vit(avg):  412, rs(sum):    0, packets: 21, drops: 13
2020-09-16T21:03:42Z [monitor] gain: 11.76, freq:  5441.5, omega: 2.589, vit(avg):  339, rs(sum):    1, packets: 19, drops: 12
2020-09-16T21:03:43Z [monitor] gain: 11.75, freq:  5441.7, omega: 2.589, vit(avg):  394, rs(sum):    3, packets: 19, drops: 13
2020-09-16T21:03:44Z [monitor] gain: 11.75, freq:  5433.0, omega: 2.589, vit(avg):  528, rs(sum):    3, packets: 18, drops: 13
2020-09-16T21:03:45Z [monitor] gain: 11.75, freq:  5446.8, omega: 2.589, vit(avg):  536, rs(sum):    4, packets: 18, drops: 13
[... cut ...]
2020-09-16T21:03:50Z [monitor] gain: 11.72, freq:  5462.7, omega: 2.589, vit(avg):  389, rs(sum):    5, packets: 19, drops: 12
2020-09-16T21:03:51Z [monitor] gain: 11.71, freq:  5444.4, omega: 2.589, vit(avg):  456, rs(sum):    0, packets: 18, drops: 13
2020-09-16T21:03:52Z [monitor] gain: 11.72, freq:  5441.3, omega: 2.589, vit(avg):  418, rs(sum):    2, packets: 19, drops: 12
2020-09-16T21:03:53Z [monitor] gain: 11.72, freq:  5452.4, omega: 2.589, vit(avg):  328, rs(sum):    0, packets: 18, drops: 13
```

hemna, 2020-09-16:
This is what I'm seeing from goesproc

```
VC 1: Detected TP_PDU drop (skipped 9 packet(s) on APID 38; prev: 2544, packet: 2554)
VC 1: VCDU drop! (lost 3; prev: 6576364; packet: 6576368)
VC 1: Detected TP_PDU drop (skipped 9 packet(s) on APID 38; prev: 2554, packet: 2564)
VC 1: VCDU drop! (lost 3; prev: 6576369; packet: 6576373)
VC 1: Detected TP_PDU drop (skipped 9 packet(s) on APID 38; prev: 2567, packet: 2577)
VC 1: VCDU drop! (lost 3; prev: 6576373; packet: 6576377)
VC 1: Detected TP_PDU drop (skipped 9 packet(s) on APID 38; prev: 2577, packet: 2587)
VC 1: VCDU drop! (lost 3; prev: 6576377; packet: 6576381)
VC 1: Detected TP_PDU drop (skipped 9 packet(s) on APID 38; prev: 2587, packet: 2597)
VC 1: VCDU drop! (lost 3; prev: 6576382; packet: 6576386)
VC 1: Detected TP_PDU drop (skipped 8 packet(s) on APID 38; prev: 2600, packet: 2609)
VC 1: VCDU drop! (lost 3; prev: 6576386; packet: 6576390)
VC 1: Detected TP_PDU drop (skipped 8 packet(s) on APID 38; prev: 2610, packet: 2619)
VC 1: VCDU drop! (lost 2; prev: 6576391; packet: 6576394)
VC 1: Detected TP_PDU drop (skipped 5 packet(s) on APID 38; prev: 2622, packet: 2628)
VC 1: VCDU drop! (lost 3; prev: 6576394; packet: 6576398)
VC 1: Detected TP_PDU drop (skipped 9 packet(s) on APID 38; prev: 2629, packet: 2639)
VC 1: VCDU drop! (lost 3; prev: 6576399; packet: 6576403)
VC 1: Detected TP_PDU drop (skipped 9 packet(s) on APID 38; prev: 2642, packet: 2652)
VC 1: VCDU drop! (lost 3; prev: 6576403; packet: 6576407)
VC 1: Detected TP_PDU drop (skipped 9 packet(s) on APID 38; prev: 2653, packet: 2663)
VC 1: VCDU drop! (lost 3; prev: 6576408; packet: 6576412)
VC 1: Detected TP_PDU drop (skipped 10 packet(s) on APID 38; prev: 2666, packet: 2677)
VC 1: VCDU drop! (lost 3; prev: 6576412; packet: 6576416)
VC 1: Detected TP_PDU drop (skipped 9 packet(s) on APID 38; prev: 2678, packet: 2688)
VC 1: VCDU drop! (lost 3; prev: 6576417; packet: 6576421)
VC 1: Detected TP_PDU drop (skipped 9 packet(s) on APID 38; prev: 2691, packet: 2701)
VC 1: VCDU drop! (lost 3; prev: 6576421; packet: 6576425)
```

pietern, 2020-09-19:
It's likely that your machine is dropping I/Q samples before they're reaching goesrecv. If you're using a Raspberry Pi, then it might have its CPU throttled. This can happen because of high temperature or high load and can be undone by rebooting it.

hemna, 2020-11-15:
It was due to a bad power supply, causing low voltage

closed 2020-11-15 by hemna