pietern/goestools #98: [QUESTION] E4000 tuner support?

An0nymo0se, 2020-11-02:
For some reason, I keep getting a Viterbi of around 2200. I spent hours trying to find the problem and I realized the SDR I used has an E4000 tuner. In the guide, it said the SDR had to use an R820T. Has that changed? It seems to recognize the tuner but the frequency fluctuates a lot (can't seem to get a lock). Any help would be much appreciated.
**_Extra info_**
**Setup:**
Antenna ----- sawbird+ goes -------- LaNA -------- NESDR Smart XTR ------- Pi
**Goesrecv logs:**
Gain: ~50db      Freq: fluctuates between -2000 and 2000      omega: ~8     vit(avg): ~2250    packets: 0       drops: ~50

Saberhawk09, 2020-11-21:
Can't say much for your problem, but I don't think it's the sdr. I use a Nooelec NESDr SMARtee XDR and it works very well. That particular SDR does have an e4000 tuner.

bt35, 2021-05-07:
Hey An0nymo0se. I have been troubleshooting this same issue on my own setup. Without success so far. Please allow me to complain a little. I really want this biastee setup to work! The USB Ports on the sawbird LNAs are not very good and mine recently broke right off the board. 

I talked with nooelec support team about the signal issue and they were very helpful. They suggested that I try setting offset_tune=1 in my rtl configuration to account for the center frequency spike that the smartee XTR has. All you need to do verify the spike exists is use the smartee in GQRX with high gain. It is a huge spike and it can obscure signals close to center. While it is valuable to use a lower gain removing the center spike is helpful. 

With the above said I have not had any good results trying to modify goesrecv to enable offset_tune in rtlsdr_source.cc and if I have read the comments on the code properly, offset_tune is disabled. This leads me to believe the e4000 is not supported or at least those who have a working setup have an antenna with better SNR, or have changed their frequency to be offset in config. (If anyone has done so... what freq settings were used?) 

I have verified my sawbird is good with a 5 volt power supply (before the USB port broke) and an airspy mini and found the bird. But no matter what I do to make the smartee XTR work it will not find the bird with goesrecv. But like I said above I can find the signal using GQRX with offset_tune=1. 

If I can figure out how to enable that in goesrecv I will follow up. Also if the issue turns out to be hardware related... I'll follow up too. I know how difficult it can be to find people with similar issues in this hobby.

bt35, 2021-05-31:
Hey hey. I'm finally revisiting this. 

I modified the code to allow for offset_tune=1 with the XTR but I still did not receive any images. I'm giving up. I just do not have a good baseline to work with here and when I think I have one, the results of the next experiment do not follow logically. In short, I think the issue is user error but I cannot isolate the fault due to blindness. Another way to say this is, like when you look for your keys and cannot see them even though they are on the kitchen table, which you looked at 10 times already. That kind of blindness. 

I keep thinking the issue could be hardware or it could be antenna but whenever I go back and look at the signal with GQRX I am convinced the issue is software. See below: 

[image]

I did have to turn up the LNA on the XTR though and I do not see how one does that in goesrecv conf file. That could be a problem too, but I doubt it. 

An Airspy mini works fine and is my baseline receiver. GQRX shows good signal too as seen below. I can regularly get 200 Vit when the antenna system is cool. 

[image]

I have seen numerous reports on other parts of the internet of others using the XTR just fine. If you are reading this and have had luck with the XTR what did you do? 

Thanks and Good Luck.

bensimms1, 2021-08-22:
It just worked for me, I plugged it in, got vit 80 and it started decoding

bt35, 2021-09-04:
Thanks for following up... and here's my follow up too. 

I suspect I have a bad solder connection or the input conductor does not make solid contact with the output conductor of the SAWBird filter. At one point the slightest movement of the XTR would cause the power light on the filter to turn off and the spectrum would drop in power, as expected when the filter/LNA has no power. 

The XTR I have is not totally broken it still receives but I do not trust my baseline signal tests anymore since I do not really know what a good signal with an XTR looks like. 

I'm going to get another XTR to try with.

BenS12362, 2021-10-16:
This is the signal I get with my setup
[image]
:

bt35, 2021-11-29:
I'll try changing the max deviation when I have time and good weather. 

I was able to get some packets the last time I tried and I remember the vit would go up. I suspect this recommendation will help.

bt35, 2022-11-20:
Long time away from this issue. But I finally figured out my problem. Read back to see my issues getting below 1000 vit-avg with the elonics 4000 SmartTeeXTR. 

I was pointed at the wrong bird. I needed to be looking at GOES 16 and not GOES 15 for HRIT. I now get about 500 vit-avg with some variability with cloud cover up too 1000 vit-avg. 

Thanks to those who helped me.