pietern/goestools #100: GOES nws directory / filenames are incorrect

KiwiInNZ, 2020-11-07:
Capturing data which spanned UTC 06-NOV-2020 - 07-NOV-2020, filenames and directories for GOES 16 / 17 / Himawari all looked perfectly valid, however the nws directories / filenames did not.

Two directories were created:
```
drwxr-xr-x 2 pi pi 4096 Nov  7 22:00 1970-01-01
drwxr-xr-x 2 pi pi 4096 Nov  7 14:00 2020-03-10
```
The first looks to be aligned with the start of Unix time and the second is from over 6 months ago (prior to starting to capture from GOES 17). So there appears to be two issues here:

- date can be set to the start date of the Unix time epoch
- date can be set to an incorrect date

Linked to this is that the filenames in each directory appear to be invalid, potentially as a result of the directory name issues.

Example filenames from the 1970-01-01 directory, with the same pattern seen for the other directory:
```
-rw-r--r-- 1 pi pi 44100 Nov  7 17:00 19700101T000000Z_20201107040005-pacsfc24_latestBW.gif
-rw-r--r-- 1 pi pi 55748 Nov  7 17:00 19700101T000000Z_20201107040045-pac24_latestBW.gif
```
The first part looks to be linked to midnight UTC for the date used for the directory filename. I'm not sure if this should be part of the filename since the actual date / time is the next part of the filename. Can this first part be removed, along with the "_" separator"? In which case the Z time zone indicator should probably be moved to the date / time in the second field.

Boosted09Foci, 2021-08-24:
I'm seeing the same issue.  Has an update to the script been made?

pietern, 2022-03-15:
The code that tries to extract a timestamp from the NWS filename is located here:
https://github.com/pietern/goestools/blob/9ca85c81784bc3b1f7e5ff409c4133eb16a3116a/src/goesproc/handler_nws_image.cc#L94-L102

Per the comment I wrote, I assume that for the HRIT stream we end up executing `parseIrregularTime`:
https://github.com/pietern/goestools/blob/9ca85c81784bc3b1f7e5ff409c4133eb16a3116a/src/goesproc/handler_nws_image.cc#L9-L71

The comment at the top here describes the file format that was used at the time I wrote this code. This format consists of the year, the month, and then the day of the year, instead of the day of the month.

@KiwiInNZ @Boosted09Foci @Mopar44084 @creinemann Could you share what the current state of things is for the NWS images that get written? The second part of the filename is what is received. The first part of the filename is what is parsed. If the second part now uses the day of month instead of day of year then the code needs to be updated to reflect this.

If you have a couple examples from February and March we can infer the pattern that is used today.

Thanks in advance.

KiwiInNZ, 2022-03-16:
I'm "fixing" the filenames automatically using https://github.com/wxcapture/wxcapture/blob/master/goes-code/wxcapture/process/find_files.py (lines 804-831) (so you can see exactly what I'm doing to fix the files).

These are downloaded from GOES 17. It would be good to get some GOES 16 ones too.

What I've done is to turn off this code (commented out line 1016) so I can see exactly what the current files are without my automated fixes.

I'll post here again later, but I expect that nothing has changed since my original logging of this issue as my fix code is still working as intended.

KiwiInNZ, 2022-03-16:
After a few hours, the following files have been created:
- 20220116T050000Z_20220316050045-hiwind_pac_latest.gif
- 20220116T050000Z_20220316050045-pac48_latestBW.gif
- 20220116T050000Z_20220316050045-WATL_latest.gif
- 20220116T050000Z_20220316050045-pac24_latestBW.gif
- 20220116T050000Z_20220316050045-pac48per_latestBW.gif

These were all created in the nws/2022-01-16 directory, with no sign of a 1970-01-01 directory being created (yet?).

This is using the following version:
```
goesproc -- 543ece3 (Tue, 5 May 2020 10:42:24 +0200)

Part of goestools (https://github.com/pietern/goestools)
Written by Pieter Noordhuis and contributors
```
Let me know if you need any additional info. I'll keep the non-correction code commented out a bit more to see if that 1970 directory gets created again.

Shout out with any more questions.

KiwiInNZ, 2022-03-20:
[files.txt](https://github.com/pietern/goestools/files/8310431/files.txt)
811 example filenames from today, with no examples of any "from" 1970, i.e. Unix epoch time start.

closed 2022-03-27 by pietern

pietern, 2022-03-27:
Thanks for the information, @KiwiInNZ.

I suspect that `1970-01-01` will show up again in April (see the branch in the logic for `month >= 4`.

I made a fix and verified it works for files that came down today. Since the pattern appears to be fixed length in all cases (the November 2020 examples and the more recent ones), I expect the fix to be conclusive.

If you update your receiver everything should work out of the box. No need to change config files.

pietern, 2022-03-27:
Many thanks to @creinemann for sharing his receiver over Tailscale so I could verify the fix.

PR #137 (2022-03-27): Fix NWS image timestamp extraction
The pattern that encodes the timestamp for NWS images used to be irregular and
required careful parsing. This pattern was changed some time before November
2020 and became regular. The irregular parser produced invalid timestamps when
applied to filenames with the new pattern. This commit updates the function
that extracts a timestamp from the filename to expect the regular pattern.

This commit also removes the timestamp prefix from the filename if it could be
extracted, allowing users to specify their own timestamp pattern.

This means that a handler with t [... cut]
commit 9f21e52d (2022-03-27): Fix NWS image timestamp extraction (#137)

The pattern that encodes the timestamp for NWS images used to be irregular and
required careful parsing. This pattern was changed some time before November
2020 and became regular. The irregular parser produced invalid timestamps when
applied to filenames with the new pattern. This commit updates the function
that extracts a timestamp from the filename to expect the regular pattern.

This commit also removes the timestamp prefix from the filename if it could be
extracted, allowing users to specify their own timestamp pattern.

This means that a handle [... cut]