pietern/goestools #99: Getting /EMWIN but not /text

rpdale, 2020-11-07:
EMWIN is just a subset of NWS products, and I was under the impression that the whole suite of files was available. While I'm seeing G16 and EMWIN directories (after adding the EMWIN configuration to the .conf file) but /text remains empty. I see the NWS text settings in the file. Is there something else I need to do?

unixpunk, 2020-11-08:
I also noticed I only ever received the text product one time in a week or so timeframe.  I think the only think in there was a single file with these contents:

`The HRIT/EMWIN broadcast utilizes virtual channels for the transmission of
several simultaneous data streams. The table below identifies the current
virtual channels assignments. The EMWIN and DCS virtual channels are permanent
assignments; the remaining channel assignment may be modified in the future.

    HRIT/EMWIN
Virtual Channel ID     Group          Product Name
------------------    -------    -------------------------------
        0             Imagery    Admin Text Messages
        1             Imagery    Mesoscale (ch. 2, 7, 13)
        2             Imagery    Band 2 - Red
        6             Imagery    GOES-15
        7             Imagery    Band 7 - Shortwave Window
        8             Imagery    Band 8
        9             Imagery    Band 9 - Mid-Level Trop
       13             Imagery    Band 13
       14             Imagery    Band 14 - IR
       15             Imagery    Band 15
       20             EMWIN      Priority
       21             EMWIN      Graphics
       22             EMWIN      Other
       23             Imagery    NWS Products
       24             Imagery    NHC Graphics Products
       25             Imagery    GOES-R JPG Products
       26             Imagery    International Graphics Products
       30             DCS        DCS Admin
       31             DCS        DCS Data
       60             Imagery    Himawari

For additional information or questions please visit
http://www.noaasis.noaa.gov/ or contact the
HRIT/EMWIN Broadcast Manager at seth.clevenstine@noaa.gov (301-817- 4558).`

rpdale, 2020-11-08:
Thanks for the note! After digging through some old documents I realized
AWIPS wasn’t part of the feed – but the AWIPS info in the decoder template
threw me 😊 Here’s the current list for GOES16 channels.

   - Rob

*From:* unixpunk <notifications@github.com>
*Sent:* Sunday, November 8, 2020 3:09 PM
*To:* pietern/goestools <goestools@noreply.github.com>
*Cc:* rpdale <rdale@skywatch.org>; Author <author@noreply.github.com>
*Subject:* Re: [pietern/goestools] Getting /EMWIN but not /text (#99)

I also noticed I only ever received the text product one time in a week or
so timeframe. I think the only think in there was a single file with these
contents:

`The HRIT/EMWIN broadcast utilizes virtual channels for the transmission of
several simultaneous data streams. The table below identifies the current
virtual channels assignments. The EMWIN and DCS virtual channels are
permanent
assignments; the remaining channel assignment may be modified in the future.

HRIT/EMWIN

Virtual Channel ID Group Product Name
------------------------------

    0             Imagery    Admin Text Messages

    1             Imagery    Mesoscale (ch. 2, 7, 13)

    2             Imagery    Band 2 - Red

    6             Imagery    GOES-15

    7             Imagery    Band 7 - Shortwave Window

    8             Imagery    Band 8

    9             Imagery    Band 9 - Mid-Level Trop

   13             Imagery    Band 13

   14             Imagery    Band 14 - IR

   15             Imagery    Band 15

   20             EMWIN      Priority

   21             EMWIN      Graphics

   22             EMWIN      Other

   23             Imagery    NWS Products

   24             Imagery    NHC Graphics Products

   25             Imagery    GOES-R JPG Products

   26             Imagery    International Graphics Products

   30             DCS        DCS Admin

   31             DCS        DCS Data

   60             Imagery    Himawari

For additional information or questions please visit
http://www.noaasis.noaa.gov/ or contact the
HRIT/EMWIN Broadcast Manager at seth.clevenstine@noaa.gov (301-817- 4558).`

—
<https://github.com/pietern/goestools/issues/99#issuecomment-723659177>, or
unsubscribe