pietern/goestools #104: Not getting admin text messages

Saberhawk09, 2020-11-21:
It's been quite a while since I used my decoding setup, and I remembered admin text messages where a thing. I had never gotten one, and was wondering if the satellites don't send them out anymore or can I enable decoding them somehow like EMWIN. Sorry if this is not the correct place, I rarely use GitHub.

agrif, 2020-12-07:
I've also stopped receiving any text products, even though my config file (the default installed by `make install`) seems to include them. I took my receiver down about a year ago and just put it back up, and something in between broke. Anybody know what's up?

spinomaly, 2021-01-01:
I believe it might be due to a format change on the filename.  It doesn't seem to include the timestamp anymore.  TextHandler is expecting a timestamp in the filename and if it doesn't exist it exits without generating the file.  A fix seems fairly straightforward depending on the requirement.  Not sure if Pieter is actively monitoring.  I'm happy to contribute.

Disclaimer: I have only been a user for around 2 weeks.  I have the code loaded into VS2017 with it being configured to compile and run the code remotely on a RPI4 allowing me to develop/debug in VS.

pietern, 2021-01-13:
Thanks for posting! And thanks for the fix, @spinomaly.

Any chance someone could send over an LRIT file of a text message? Perhaps there still is a timestamp in there somewhere, just encoded in a different way. If not, I'm happy to merge the PR that uses the current local time.

Including a time in the filename seems like the right thing to do, in case the contents of this file ever changes.

Saberhawk09, 2021-01-14:
> Thanks for posting! And thanks for the fix, @spinomaly.
>
> Any chance someone could send over an LRIT file of a text message? Perhaps there still is a timestamp in there somewhere, just encoded in a [...]

Thanks so much for the help guys! My decoding station is currently out of commission, but when I eventually fix it I'll try this.

spinomaly, 2021-01-14:
Welcome!  Attached is a text LRIT file that just occured.

[GOES_WEST_Admin_message_updated_2020.11.23.zip](https://github.com/pietern/goestools/files/5812130/GOES_WEST_Admin_message_updated_2020.11.23.zip)

Saberhawk09, 2021-01-14:
> Welcome! Attached is a text LRIT file that just occured.
> 
> [GOES_WEST_Admin_message_updated_2020.11.23.zip](https://github.com/pietern/goestools/files/5812130/GOES_WEST_Admin_message_updated_2020.11.23.zip)

Awesome, now how would someone go about updating their config to RX these by default? Also can I decode this LRIT file manually with goesproc?

spinomaly, 2021-01-14:
I don't think goesproc will process this file.  Same issue whether streaming or loading files.  However, I have never tried processing files.  I am only processing streams at the moment.

Saberhawk09, 2021-01-14:
> I don't think goesproc will process this file.  Same issue whether streaming or loading files.  However, I have never tried processing files.  I am only processing streams at the moment.

Ok then, but am I still able to modify the goesrecv/goesproc config to enable reception of admin text?

spinomaly, 2021-01-14:
Add this to goesproc config file.  However, no files will be saved until a fix is implemented.

[[handler]]
type = "text"
origin = "other"
directory = "./text/{time:%Y-%m-%d}"
filename = "{filename}_{time:%Y%m%dT%H%M%SZ}"

Saberhawk09, 2021-01-14:
Ah ok, so this is preemptive until a patch can be made? Like I said, thanks for your help I really appreciate it. These kind of things just bug me lol.

spinomaly, 2021-01-19:
If you want to use a fix I implemented to enable saving of admin messages you can get the latest in my repo.