pietern/goestools #111: Stat: File or directory not found goesproc

zeke800, 2021-08-05:
Hi

This is a wonderful piece of software, but I am getting one error on Raspberry pi 3. I am getting from goesproc, when I run the command goesproc -c /usr/share/goestools/goesproc-goesr.conf -m packet – subscribe tcp://127.0.0.1:5004 – out /home/pi/goes, I  get the error "stat: no such file or directory. I have no 'stat' file anywhere, and I have no idea what this means. I am getting around 100 vit avg and the demodulator works perfectly. Config for demodulator: 

[demodulator]
mode = "hrit"
source = "rtlsdr"

[rtlsdr]
frequency = 1694100000
sample_rate = 2000000
gain = 42
bias_tee = true

[costas]
max_deviation = 200e3

[decoder.packet_publisher]
bind = "tcp://0.0.0.0:5004"
send_buffer = 1048576

[monitor]
statsd_address = "udp4://localhost:8125"
For decoder:
# Example configuration file for goesproc
#
# This tool is designed to run on streaming data (live or recorded)
# and product whatever is listed in this file. A single product can be
# processed multiple times (e.g. with different contrast curves,
# different scale, or different annotations) by listing multiple
# handlers for that same product.
#

# GOES-16 mesoscale region 1 imagery is stored at ./goes16/m1/YYYY-MM-DD
# The pattern specified in {time:XXX} is extrapolated using strftime(3).
# It can be used more than once if needed.
[[handler]]
type = "image"
origin = "goes16"
region = "m1"
dir = "./goes16/m1/{time:%Y-%m-%d}"

# GOES-16 full disk originals.
[[handler]]
type = "image"
origin = "goes16"
region = "fd"
dir = "./goes16/fd/{time:%Y-%m-%d}"

# GOES-16 full disk, channel 2, with contrast curve applied.
# The section [handler.remap] below applies to this handler.
[[handler]]
type = "image"
origin = "goes16"
region = "fd"
channels = [ "ch02" ]
directory = "./goes16/fd/{time:%Y-%m-%d}"
filename = "{filename}_contrast"

  # Check out https://github.com/hdoverobinson/wx-star_false-color for a
  # GOES-16 contrast curve and false color LUT. Uncomment the snippet
  # below and update the path to point to the contrast curve file.

# GOES-16 full disk, channel 2 and 13 to produce a false color image.
# The sections [handler.remap] and [handler.lut] below apply to this handler.
[[handler]]
type = "image"
origin = "goes16"
region = "fd"
channels = [ "ch02", "ch13" ]
directory = "./goes16/false-color/fd/{time:%Y-%m-%d}"

  

# The raw GOES-15 full disk images are a mix of left
# and right aligned images. The image headers contain
# navigational data and we can use this to produce
# identical crops regardless of alignment.
[[handler]]
type = "image"
origin = "goes15"
region = "fd"
crop = [ -2373, 2371, -1357, 1347 ]
directory = "./goes15/fd/{time:%Y-%m-%d}"
filename = "GOES15_{region:short}_{channel:short}_{time:%Y%m%dT%H%M%SZ}"

# The HRIT stream on the GOES-R series includes EMWIN data.
# This handler writes EMWIN files and decompresses them if applicable
# (a subset of EMWIN files is compressed as ZIP file).
[[handler]]
type = "emwin"
directory = "./emwin/{time:%Y-%m-%d}"

# NWS text messages
#
# The AWIPS Product Identifier of the message can be used in the
# filename pattern. See https://www.weather.gov/tg/awips for a
# description of the fields in this identifier. They are usable
# through the following keywords (in "{awips:...}"):
#
#  * t1t2
#  * a1a2
#  * ii
#  * cccc
#  * yy
#  * gggg
#  * bbb
#  * nnn
#  * xxx
#
[[handler]]
type = "text"
origin = "nws"
directory = "./nws/{time:%Y-%m-%d}"
filename = "{time:%Y%m%dT%H%M%SZ}_{awips:nnn}{awips:xxx}"

# NWS images
#
# Files received through HRIT do not have an LRIT timestamp header and
# goesproc tries to parse their timestamp from their file name. This
# format is not specified anywhere so it might change at any point in
# time. If the timestamp cannot be parsed from the file name, goesproc
# defaults to 1970-01-01. @pietern pinged NOAA (Seth) about this.
#
# NWS image files received through LRIT (GOES-15) have a timestamp.
#
[[handler]]
type = "image"
origin = "nws"
directory = "./nws/{time:%Y-%m-%d}"
filename = "{time:%Y%m%dT%H%M%SZ}_{filename}"

# Miscellaneous text.
[[handler]]
type = "text"
origin = "other"
directory = "./text/{time:%Y-%m-%d}"
filename = "{time:%Y%m%dT%H%M%SZ}_{filename}"
How can I solve this issue? It the decoder worked fine before the pi crashed. Thanks in advance!!

zeke800, 2021-08-05:
Oops! The config again: 
`# Example configuration file for goesproc
#
# This tool is designed to run on streaming data (live or recorded)
# and product whatever is listed in this file. A single product can be
# processed multiple times (e.g. with different contrast curves,
# different scale, or different annotations) by listing multiple
# handlers for that same product.
#

# GOES-16 mesoscale region 1 imagery is stored at ./goes16/m1/YYYY-MM-DD
# The pattern specified in {time:XXX} is extrapolated using strftime(3).
# It can be used more than once if needed.
[[handler]]
type = "image"
origin = "goes16"
region = "m1"
dir = "./goes16/m1/{time:%Y-%m-%d}"

# GOES-16 full disk originals.
[[handler]]
type = "image"
origin = "goes16"
region = "fd"
dir = "./goes16/fd/{time:%Y-%m-%d}"

# GOES-16 full disk, channel 2, with contrast curve applied.
# The section [handler.remap] below applies to this handler.
[[handler]]
type = "image"
origin = "goes16"
region = "fd"
channels = [ "ch02" ]
directory = "./goes16/fd/{time:%Y-%m-%d}"
filename = "{filename}_contrast"

  # Check out https://github.com/hdoverobinson/wx-star_false-color for a
  # GOES-16 contrast curve and false color LUT. Uncomment the snippet
  # below and update the path to point to the contrast curve file.

# GOES-16 full disk, channel 2 and 13 to produce a false color image.
# The sections [handler.remap] and [handler.lut] below apply to this handler.
[[handler]]
type = "image"
origin = "goes16"
region = "fd"
channels = [ "ch02", "ch13" ]
directory = "./goes16/false-color/fd/{time:%Y-%m-%d}"

  

# The raw GOES-15 full disk images are a mix of left
# and right aligned images. The image headers contain
# navigational data and we can use this to produce
# identical crops regardless of alignment.
[[handler]]
type = "image"
origin = "goes15"
region = "fd"
crop = [ -2373, 2371, -1357, 1347 ]
directory = "./goes15/fd/{time:%Y-%m-%d}"
filename = "GOES15_{region:short}_{channel:short}_{time:%Y%m%dT%H%M%SZ}"

# The HRIT stream on the GOES-R series includes EMWIN data.
# This handler writes EMWIN files and decompresses them if applicable
# (a subset of EMWIN files is compressed as ZIP file).
[[handler]]
type = "emwin"
directory = "./emwin/{time:%Y-%m-%d}"

# NWS text messages
#
# The AWIPS Product Identifier of the message can be used in the
# filename pattern. See https://www.weather.gov/tg/awips for a
# description of the fields in this identifier. They are usable
# through the following keywords (in "{awips:...}"):
#
#  * t1t2
#  * a1a2
#  * ii
#  * cccc
#  * yy
#  * gggg
#  * bbb
#  * nnn
#  * xxx
#
[[handler]]
type = "text"
origin = "nws"
directory = "./nws/{time:%Y-%m-%d}"
filename = "{time:%Y%m%dT%H%M%SZ}_{awips:nnn}{awips:xxx}"

# NWS images
#
# Files received through HRIT do not have an LRIT timestamp header and
# goesproc tries to parse their timestamp from their file name. This
# format is not specified anywhere so it might change at any point in
# time. If the timestamp cannot be parsed from the file name, goesproc
# defaults to 1970-01-01. @pietern pinged NOAA (Seth) about this.
#
# NWS image files received through LRIT (GOES-15) have a timestamp.
#
[[handler]]
type = "image"
origin = "nws"
directory = "./nws/{time:%Y-%m-%d}"
filename = "{time:%Y%m%dT%H%M%SZ}_{filename}"

# Miscellaneous text.
[[handler]]
type = "text"
origin = "other"
directory = "./text/{time:%Y-%m-%d}"
filename = "{time:%Y%m%dT%H%M%SZ}_{filename}"
`

Volas171, 2022-01-24:
I had the same issue, solved it by not using -

log into your pi
create a goes folder 
`mkdir goes && cd goes`
then open a screen session and execute goesproc
`screen -mS goesproc`
(inside the screen)
` goesproc --config /usr/share/goestools/goesproc-goesr.conf --mode packet --subscribe tcp://127.0.0.1:5004`
that should fix the issue and store everything into /home/pi/goes