pietern/goestools #113: Some issues building with Buster

RyLeeRyno, 2021-08-20:
Hello Goesers,

I'm new to Github so forgive me if I'm not providing build recommendations in the correct context.  

To Summarize:  Debian Bullseye introduced OpenCV 4 and Project 7 which are incompatible with the existing build.  I've had to make a few modifications as follows:

CMakeLists.txt in the main folder.
 - Line 84 replace with:  **include_directories(BEFORE SYSTEM /usr/include/opencv4)**

src/goesproc/CMakeLists.txt
 - Line 23 replace with:  **pkg_check_modules(OPENCV REQUIRED opencv4)**

src/lib/packet_writer.h
 - Line 5 add: **#include <stdint.h>**
 
src/lib/packet_writer.h
 - Line 4 add: **#include <stdint.h>**

src/goesproc/config.cc
 - Line 210 replace with:  **auto img = cv::imread(path, cv::IMREAD_UNCHANGED);**

src/goesproc/image.cc
 - Lines 302 and 309 replace:  return std::move(out); with **return out;**

I hope this is helpful for anyone struggling with compiling goestools with Bullseye.
Thank you.

hjf, 2021-11-02:
I added those changes but I'm getting this error on the latest Bullseye:

```[ 62%] Building CXX object src/lib/CMakeFiles/packet_reader.dir/packet_reader.cc.o
In file included from /home/pi/goestools/src/lib/packet_reader.cc:1:
/home/pi/goestools/src/lib/packet_reader.h:11:38: error: ‘uint8_t’ was not declared in this scope
   11 |   virtual bool nextPacket(std::array<uint8_t, 892>& out) = 0;
      |                                      ^~~~~~~
/home/pi/goestools/src/lib/packet_reader.h:11:50: error: template argument 1 is invalid
   11 |   virtual bool nextPacket(std::array<uint8_t, 892>& out) = 0;
      |                                                  ^
make[2]: *** [src/lib/CMakeFiles/packet_reader.dir/build.make:82: src/lib/CMakeFiles/packet_reader.dir/packet_reader.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1681: src/lib/CMakeFiles/packet_reader.dir/all] Error 2
```

Higgy710, 2021-12-17:
`CMakeLists.txt in the main folder.

    Line 84 replace with: include_directories(BEFORE SYSTEM /usr/include/opencv4)
`

Replacing this messed up some dependencies.
Adding this line after 84 worked though.