pietern/goestools #114: goesproc: Assertion 'rv == 0' failed

reykowski, 2021-09-02:
Hi Pieter,
I helped a friend of mine to set up goes tools on an R-Pi. 
He had his setup running for over 6 months, then something strange was happening.
After starting goesproc, the software was running for several minutes, displaying
mkdir: Permission deniedcounter=15492799
Then the code terminated with the following message:
Assertion 'rv == 0' failed at /home/pi/goestools/src/util/fs.cc:23
I think it is also important to mention that he uses an external USB drive to store the data.
Thanks in advance,
Arne

ThePhotoChemist, 2022-08-20:
I'm not 100% sure if this is the answer, but I'm hoping this helps someone.  

I had a similar issue.  I had a setup that ran for about 8 months.  At some point the computer shut down, and I was receiving this error message.  I wasn't able to figure it out, updated Ubuntu, couldn't compile, etc etc and I just let it sit idle for another 6 months.  

Recently I did a fresh install and got everything working again... until I got this same error message after a few minutes.  But this time I realized, the way I had my script setup, goesproc was trying to write the files to whatever my current directory was... which at the time, was root.  Goesproc would run happily long enough until it needed to write, and then it would crash.

Once I changed back to my home directory and made sure the goes16 directory had adequate permissions, I haven't had any issues since.

mbruski, 2022-12-27:
I had this issue last evening when out of the blue I received the same error.  Turned out the partition I was saving all my files and images to had saturated.  After moving a bunch of stuff off to another drive I was back in business.