pietern/goestools #117: Compatibility with Zero 2 W

funtime1818, 2021-10-30:
It is mentioned in the Readme that the RX pipeline can run on a Raspberry Pi 3, and I was wondering if there would be any difficulty getting it running on a Zero 2 W, as the performance is slightly lower. Thank you!

porkiedev, 2022-01-18:
I haven't actually been able to test out goestools for myself yet so I can't speak for software compatability (though I imagine it would work just fine)

But for the hardware specs, The Z2W's CPU is rated for 1GHz. The Pi3 is 1.2GHz. There is always the option of running a 100-200MHz overclock on the Z2W provided you can keep it cool (I would suggest thermal paste or a heatsink case, thermal pads suck)
People have overclocked the Z2W to 1.4GHz and had stable results. I have no doubt a 1.2GHz would work fine.

That said, if goestools is memory intensive I wouldn't particularly suggest it, the Z2W only has 0.5GB of ram as opposed to the Pi3's 1GB.

DakotaNelson, 2023-01-06:
'goesproc' doesn't work on a raspi zero, because OpenCV won't compile on it (due to lack of RAM) - but building just `goesrecv` using 'cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local -DBUILD_GOESPROC=OFF` works at least to compile.

I'm running into an unrelated issue right now (#155) that's preventing me from testing it running, but I suspect the zero should be able to receive packets, you'll just have to process them elsewhere.