pietern/goestools #121: Compiling on Ubuntu

JosephTheTank13, 2022-01-27:
I have tried many times over the years to compile this on my main Ubuntu PC and have yet to succeed. On a raspberry pi it usually works first try, but there is a difference in how cmake works or something.

`~/goestools/build$ cmake ../ -DCMAKE_INSTALL_PREFIX=/usr/local
CMake Warning at CMakeLists.txt:36 (find_package):
  By not providing "FindSanitizers.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Sanitizers", but CMake did not find one.

  Could not find a package configuration file provided by "Sanitizers" with
  any of the following names:

    SanitizersConfig.cmake
    sanitizers-config.cmake

  Add the installation prefix of "Sanitizers" to CMAKE_PREFIX_PATH or set
  "Sanitizers_DIR" to a directory containing one of the above files.  If
  "Sanitizers" provides a separate development package or SDK, be sure it has
  been installed.

CMake Error at CMakeLists.txt:59 (add_subdirectory):
  The source directory

    /home/joseph/goestools/vendor/libcorrect

  does not contain a CMakeLists.txt file.

CMake Error at CMakeLists.txt:60 (add_subdirectory):
  The source directory

    /home/joseph/goestools/vendor/libaec

  does not contain a CMakeLists.txt file.

CMake Error at CMakeLists.txt:68 (add_subdirectory):
  The source directory

    /home/joseph/goestools/vendor/nanomsg

  does not contain a CMakeLists.txt file.

CMake Error at src/lrit/CMakeLists.txt:5 (add_sanitizers):
  Unknown CMake command "add_sanitizers".

-- Configuring incomplete, errors occurred!
See also "/home/joseph/goestools/build/CMakeFiles/CMakeOutput.log".`

I am not versed in how CMake works and how to go about fixing this so if anyone at all knows what's wrong here.... I'd really really be grateful.

MustBeArt, 2022-01-27:
As it happens, I just ran into this problem earlier today.

The build instructions in the README are incomplete with respect to cloning
the repo. After the git clone,

  cd goestools
  git submodule init
  git submodule update --recursive

Then, if your experience is like mine, you can start having fun with
missing dependencies and version bit rot.
Check issue #113 in the Github repo for some patches. Also,

  sudo apt-get install libairspy-dev librtlsdr-dev libopencv-dev libproj-dev

I can't guarantee success, though, since I'm still trying to get the build
to work.

  -Paul

JosephTheTank13, 2022-01-27:
Any idea how to get OpenCV to work? I installed libopencv-dev, and changed
the lines in the CMakeLists.txt (main and goesproc) to reflect openCV 4 but
iirc that does not help at all. What errors are you running into that’s
preventing you from compiling?

JosephTheTank13, 2022-01-27:
I get as far as `[ 64%] Linking CXX static library liblrit.a
[ 64%] Built target lrit
make: *** [Makefile:171: all] Error 2
`
And of course it doesn't give any information on why it failed.

JosephTheTank13, 2022-01-27:
[ 69%] Built target compute_sync_words
[ 71%] Built target airspy_source
[ 72%] Built target assembler
[ 73%] Built target nanomsg_source
[ 75%] Built target agc
[ 75%] Built target quantize
In file included from /home/joseph/goestools/src/lib/packet_reader.cc:1:
/home/joseph/goestools/src/lib/packet_reader.h:11:38: error: ‘uint8_t’ was not declared in this scope
   11 |   virtual bool nextPacket(std::array<uint8_t, 892>& out) = 0;
      |                                      ^~~~~~~
/home/joseph/goestools/src/lib/packet_reader.h:4:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
    3 | #include <array>
  +++ |+#include <cstdint>
    4 | 
/home/joseph/goestools/src/lib/packet_reader.h:11:50: error: template argument 1 is invalid
   11 |   virtual bool nextPacket(std::array<uint8_t, 892>& out) = 0;
      |                                                  ^
make[2]: *** [src/lib/CMakeFiles/packet_reader.dir/build.make:82: src/lib/CMakeFiles/packet_reader.dir/packet_reader.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1768: src/lib/CMakeFiles/packet_reader.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/joseph/goestools/src/lib/packet_writer.cc:1:
/home/joseph/goestools/src/lib/packet_writer.h:12:39: error: ‘uint8_t’ was not declared in this scope
   12 |   virtual void write(const std::array<uint8_t, 892>& in, time_t t) = 0;
      |                                       ^~~~~~~
/home/joseph/goestools/src/lib/packet_writer.h:5:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
    4 | #include <ctime>
  +++ |+#include <cstdint>
    5 | 
/home/joseph/goestools/src/lib/packet_writer.h:12:51: error: template argument 1 is invalid
   12 |   virtual void write(const std::array<uint8_t, 892>& in, time_t t) = 0;
      |                                                   ^
make[2]: *** [src/lib/CMakeFiles/packet_writer.dir/build.make:82: src/lib/CMakeFiles/packet_writer.dir/packet_writer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1849: src/lib/CMakeFiles/packet_writer.dir/all] Error 2
[ 76%] Built target rtlsdr_source
[ 77%] Built target rrc
[ 77%] Built target costas
In file included from /home/joseph/goestools/src/decoder/derandomizer.cc:1:
/home/joseph/goestools/src/decoder/derandomizer.h:14:27: error: ‘size_t’ has not been declared
   14 |   void run(uint8_t* data, size_t len);
      |                           ^~~~~~
/home/joseph/goestools/src/decoder/derandomizer.cc:33:6: error: no declaration matches ‘void decoder::Derandomizer::run(uint8_t*, size_t)’
   33 | void Derandomizer::run(uint8_t* data, size_t len) {
      |      ^~~~~~~~~~~~
In file included from /home/joseph/goestools/src/decoder/derandomizer.cc:1:
/home/joseph/goestools/src/decoder/derandomizer.h:14:8: note: candidate is: ‘void decoder::Derandomizer::run(uint8_t*, int)’
   14 |   void run(uint8_t* data, size_t len);
      |        ^~~
/home/joseph/goestools/src/decoder/derandomizer.h:10:7: note: ‘class decoder::Derandomizer’ defined here
   10 | class Derandomizer {
      |       ^~~~~~~~~~~~
make[2]: *** [src/decoder/CMakeFiles/packetizer.dir/build.make:95: src/decoder/CMakeFiles/packetizer.dir/derandomizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2071: src/decoder/CMakeFiles/packetizer.dir/all] Error 2
make: *** [Makefile:171: all] Error 2

MustBeArt, 2022-01-27:
On Wed, Jan 26, 2022 at 6:31 PM JosephTheTank13 ***@***.***>
wrote:

> Any idea how to get OpenCV to work? I installed libopencv-dev, and changed
> the lines in the CMakeLists.txt (main and goesproc) to reflect openCV 4 but
> iirc that does not help at all.

I copied from another message the edits for OpenCV. In the top-level
CMakeLists.txt, add
include_directories(BEFORE SYSTEM /usr/include/opencv4)

In src/goesproc/CMakeLists.txt, change:
-pkg_check_modules(OPENCV REQUIRED opencv)
+pkg_check_modules(OPENCV REQUIRED opencv4)

In src/goesproc/config.cc, change:
-        auto img = cv::imread(path, CV_LOAD_IMAGE_UNCHANGED);
+        auto img = cv::imread(path, cv::IMREAD_UNCHANGED);

Two places in src/goesproc/image.cc, change:
-  return std::move(out);
+  return out;

I'm not sure if it helped, but I also apt-get installed
libopencv-contrib-dev.

What errors are you running into that’s preventing you from compiling?
>

I had to mess with the checks on the version of Proj. Search the web for
ACCEPT_USE_OF_DEPRECATED_PROJ_API_H
and you'll also have to short-circuit a check in one of the .h files.

Another thing that slowed me down was that I needed to greatly increase the
amount of swap space, but that's just because I was compiling on a
Raspberry Pi.

Above advice gets it to COMPILE and LINK. I don't know yet whether it WORKS!

Some of this stuff is already in a pull request that's been languishing
since August.

  -Paul

> Message ID: ***@***.***>
>

MustBeArt, 2022-01-27:
> defined in header ‘’; did you forget to ‘#include ’?

See https://github.com/pietern/goestools/issues/113 for most of those
errors.

Others, I didn't run into.

  -Paul

JosephTheTank13, 2022-01-27:
Alright, I made some changes but am still stuck.

[ 71%] Built target compute_sync_words
[ 71%] Built target sz
[ 72%] Built target airspy_source
[ 73%] Built target nanomsg_source
In file included from /home/joseph/goestools/src/decoder/derandomizer.cc:1:
/home/joseph/goestools/src/decoder/derandomizer.h:14:27: error: ‘size_t’ has not been declared
   14 |   void run(uint8_t* data, size_t len);
      |                           ^~~~~~
[ 75%] Built target quantize
[ 75%] Built target agc
[ 76%] Built target rtlsdr_source
[ 76%] Built target rrc
[ 77%] Built target costas
[ 78%] Built target clock_recovery
[ 80%] Built target assembler
[ 82%] Built target benchmark
/home/joseph/goestools/src/decoder/derandomizer.cc:33:6: error: no declaration matches ‘void decoder::Derandomizer::run(uint8_t*, size_t)’
   33 | void Derandomizer::run(uint8_t* data, size_t len) {
      |      ^~~~~~~~~~~~
In file included from /home/joseph/goestools/src/decoder/derandomizer.cc:1:
/home/joseph/goestools/src/decoder/derandomizer.h:14:8: note: candidate is: ‘void decoder::Derandomizer::run(uint8_t*, int)’
   14 |   void run(uint8_t* data, size_t len);
      |        ^~~
/home/joseph/goestools/src/decoder/derandomizer.h:10:7: note: ‘class decoder::Derandomizer’ defined here
   10 | class Derandomizer {
      |       ^~~~~~~~~~~~
make[2]: *** [src/decoder/CMakeFiles/packetizer.dir/build.make:95: src/decoder/CMakeFiles/packetizer.dir/derandomizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2071: src/decoder/CMakeFiles/packetizer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 82%] Linking CXX static library libpacket_writer.a
[ 82%] Built target packet_writer
[ 82%] Linking CXX static library libpacket_reader.a
[ 82%] Built target packet_reader
make: *** [Makefile:171: all] Error 2

MustBeArt, 2022-01-27:
> In file included from /home/joseph/goestools/src/decoder/derandomizer.cc:1:
> /home/joseph/goestools/src/decoder/derandomizer.h:14:27: error: ‘size_t’ has not been declared
> 
I didn’t run into this problem, but I bet you could make it go away by

#include <cstddef>

In derandomizer.h

I think the rest of the messages are all about the same error, so they might go away too.

  -Paul

JosephTheTank13, 2022-01-27:
Getting close! 

[ 88%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/map_drawer.cc.o
In file included from /home/joseph/goestools/src/goesproc/proj.cc:1:
/home/joseph/goestools/src/goesproc/proj.h:4:2: error: #error "proj version 4 or 5 required"
    4 | #error "proj version 4 or 5 required"
      |  ^~~~~
[ 89%] Built target packetdump
[ 92%] Built target goesrecv
In file included from /home/joseph/goestools/src/goesproc/map_drawer.h:7,
                 from /home/joseph/goestools/src/goesproc/handler_goesn.cc:12:
/home/joseph/goestools/src/goesproc/proj.h:4:2: error: #error "proj version 4 or 5 required"
    4 | #error "proj version 4 or 5 required"
      |  ^~~~~
In file included from /home/joseph/goestools/src/goesproc/map_drawer.h:7,
                 from /home/joseph/goestools/src/goesproc/map_drawer.cc:1:
/home/joseph/goestools/src/goesproc/proj.h:4:2: error: #error "proj version 4 or 5 required"
    4 | #error "proj version 4 or 5 required"
      |  ^~~~~
In file included from /home/joseph/goestools/src/goesproc/map_drawer.h:7,
                 from /home/joseph/goestools/src/goesproc/handler_himawari8.cc:12:
/home/joseph/goestools/src/goesproc/proj.h:4:2: error: #error "proj version 4 or 5 required"
    4 | #error "proj version 4 or 5 required"
      |  ^~~~~
In file included from /home/joseph/goestools/src/goesproc/map_drawer.h:7,
                 from /home/joseph/goestools/src/goesproc/handler_goesr.cc:14:
/home/joseph/goestools/src/goesproc/proj.h:4:2: error: #error "proj version 4 or 5 required"
    4 | #error "proj version 4 or 5 required"
      |  ^~~~~
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:316: src/goesproc/CMakeFiles/goesproc.dir/proj.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:329: src/goesproc/CMakeFiles/goesproc.dir/map_drawer.cc.o] Error 1
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:173: src/goesproc/CMakeFiles/goesproc.dir/handler_goesn.cc.o] Error 1
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:199: src/goesproc/CMakeFiles/goesproc.dir/handler_himawari8.cc.o] Error 1
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:186: src/goesproc/CMakeFiles/goesproc.dir/handler_goesr.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2549: src/goesproc/CMakeFiles/goesproc.dir/all] Error 2
make: *** [Makefile:171: all] Error 2

It seems I don't have the version of proj that it's looking for.

JosephTheTank13, 2022-01-27:
So it's looking for either version 4 or version 5 of libproj. The problem is Ubuntu 21.10 has libproj 7.2.1-1. I need to somehow get the old version or update goesproc to accept version 7.

JosephTheTank13, 2022-01-27:
Alright! Success at last! I am working on a guide right now with all the modifications listed.

JosephTheTank13, 2022-01-27:
Guide for fixing GOESTOOLS 2022-1-27

sudo apt-get install libairspy-dev librtlsdr-dev libopencv-dev libproj-dev cmake
git clone https://github.com/pietern/goestools
cd goestools
git submodule init
git submodule update --recursive

in the root directory
CMakeLists.txt, add
include_directories(BEFORE SYSTEM /usr/include/opencv4) to line 84

In src/goesproc/CMakeLists.txt, line 23, change:
pkg_check_modules(OPENCV REQUIRED opencv) to
pkg_check_modules(OPENCV REQUIRED opencv4)

In src/goesproc/config.cc, line 210 change:
auto img = cv::imread(path, CV_LOAD_IMAGE_UNCHANGED); to
auto img = cv::imread(path, cv::IMREAD_UNCHANGED);

Two places (line 302 and 309) in src/goesproc/image.cc, change:
return std::move(out); 
to
return out;

Add: #include <stdint.h> to:
src/lib/packet_reader.h (line 5)
src/lib/packet_writer.h (line 4)

Next add: #include <cstddef> 
to 
/src/decoder/derandomizer.cc

To the beginning of /src/goesproc/proj.cc and proj.h, add
#define ACCEPT_USE_OF_DEPRECATED_PROJ_API_H 1

And because proj has been updated a lot, change
#if PROJ_VERSION_MAJOR < 4 || PROJ_VERSION_MAJOR > 5
to
#if PROJ_VERSION_MAJOR < 4 || PROJ_VERSION_MAJOR > 7
in /src/goesproc/proj.h

mkdir -p build
cd build
cmake ../ -DCMAKE_INSTALL_PREFIX=/usr/local
sudo make -j6 install

Now we need to set up our config files

in your home directory, create goesrecv.conf
In goesrecv.conf, paste in:
"
 [demodulator]
mode = "hrit"
source = "rtlsdr"

[rtlsdr]
frequency = 1694100000
sample_rate = 2400000
gain = 5
bias_tee = false

[costas]
max_deviation = 200e3

[decoder.packet_publisher]
bind = "tcp://0.0.0.0:5004"
send_buffer = 1048576

[monitor]
statsd_address = "udp4://localhost:8125"
"
Run in a terminal window:

goesrecv -v -i 1 -c ~/goesrecv.conf

then: 
goesproc -c /usr/local/share/goestools/goesproc-goesr.conf -m packet  --subscribe tcp://127.0.0.1:5004

And that's how it's done!

closed 2022-03-15 by pietern

JosephTheTank13, 2022-03-18:
> Thanks for sharing these steps @JosephTheTank13 and @MustBeArt !
> 
> All of these are now present in the main branch so it should work out of the box again.
> 
> Also, I've setup automated tests to compile goestools against a [handful of versions of Ubuntu](https://github.com/pietern/goestools/actions/runs/1843756218) (including 21.10), confirming that compilation succeeds. This also uncovered that the libproj API usage must be updated to make it work for the upcoming Ubuntu 22.04 release.

The legend finally returns! Thank you!