pietern/goestools #122: Update for Goestools to recieve GOES-18 and Himawari-9

creinemann, 2022-02-03:
A few sections of goestools need to be updated to allow the creation of imagery files from the upcoming GOES-T (GOES-18) which should be transitting HRIT by Jan 3 2023.  
`src/goesproc/handler_goesr.cc` may need to be updated to reflect GOES-18 as well
```
  // encoded in the ancillary data field.
  //
  auto nlh = f->getHeader<lrit::NOAALRITHeader>();
  if (nlh.productID != 16 && nlh.productID != 17 && nlh.productID != 18 ) {
    return;
```

In addition to changes to the code, an additional handler in the conf file will be needed as well
an example of a portion of a "GOES-18" handler:

```
# Store all original GOES-18 products.
[[handler]]
type = "image"
origin = "goes18"
directory = "./goes18/{region:short|lower}/{channel:short|lower}/{time:%Y-%m-%d}"
filename = "GOES18_{region:short}_{channel:short}_{time:%Y%m%dT%H%M%SZ}"
format = "jpg"
json = false

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_0_countries_lakes.json"

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_1_states_provinces_lakes.json"
```
________________________________
Himawari 9 transition from Himawari 8 WAS scheduled for April 2022, but has been pushed to December 2022, Again, some updates need to be made, such as src/goesproc/handler_himawari8.cc  It is not known as of yet if the NOAA LRIT header file will remain '43' or be changed.

Again an update to the users conf file would be needed such as:
```
# Images relayed from Himawari-9 Sched to be Primary 12/2022.
[[handler]]
type = "image"
origin = "himawari9"
directory = "./himawari9/{region:short|lower}/{time:%Y-%m-%d}"
filename = "Himawari9_{region:short}_{channel:short}_{time:%Y%m%dT%H%M%SZ}"
format = "jpg"
json = false

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_0_countries_lakes.json"

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_1_states_provinces_lakes.json"
```

creinemann, 2022-02-14:
@pietern If I can help in any way, let me know.  Just not a proficient coder!

pietern, 2022-03-15:
@creinemann I can make the changes needed for GOES-18. But I'd like to confirm the `productID` for GOES-17 first. If this is 16 for all GOES-R series satellites and some other product uses 18, making the change could crash goesproc on new payloads. The actual filtering on satellite is done a bit further down: https://github.com/pietern/goestools/blob/9ca85c81784bc3b1f7e5ff409c4133eb16a3116a/src/goesproc/handler_goesr.cc#L409-L417

If you have an active GOES-17 receiver, could you capture a couple minutes worth of packets and send them to me?

The packets will include GOES-17 LRIT files for me to check their `productID`.

Thanks in advance.

pietern, 2022-03-15:
You can capture packets with:
```
goespackets --subscribe [ADDRESS] --record
```
After a few minutes you can press Ctrl-C to interrupt and stop the capture.

creinemann, 2022-03-15:
Sent soome GOES-17 packets just now via email.

Carl

pietern, 2022-03-16:
Thanks, @creinemann!

To process these, I first ran `goeslrit` on the packets:
```
goeslrit --out /tmp/lrit --all /tmp/packets*
```

The list of LRIT files include a bunch of EMWIN data, some DCS, and GOES-17 ABI L2 product files, for example:
`OR_ABI-L2-CMIPF-M6C09_G17_s20220742100321_e20220742109393_c20220742109456_000.lrit`.

To inspect that, I ran `lritdump` on that file:
```
lritdump /tmp/lrit/OR_ABI-L2-CMIPF-M6C09_G17_s20220742100321_e20220742109393_c20220742109456_000.lrit
```

And got the following output: https://gist.github.com/pietern/474b1efaf9cb2febbad59a62838ab237

What I was looking for is the `ProductID` under the `NOAA LRIT` header. The value here is 17 and that implies (to me) that NOAA sets this to be equal to the satellite ID. This in turn means that 18 and 19 need to be added to the list of allowed product IDs. This first filtering step happens early to reject all LRIT files that are not GOES-R series image data.

The second filtering step filters on a specific satellite, product, region, and channel. This step uses the information in the ancillary text header (where, for example, the satellite is specified as `Satellite = G17;`).

commit d61405e2 (2022-03-16): Make GOES-R handler work for GOES-18 and GOES-19 (#135)

Closes #122.
closed 2022-03-16 by pietern