pietern/goestools #130: Upgrade proj bindings

pietern, 2022-02-14:
The proj API has changed in version 5 (see https://proj.org/development/migration.html).

Backwards compatibility works until version 7.

Ubuntu 22.04 will ship with proj version 8 where this no longer works (see [compilation failure](https://github.com/pietern/goestools/runs/5191508718?check_suite_focus=true)).

Both bindings can coexist and used based on `PROJ_VERSION_MAJOR`.

soxrok2212, 2022-05-13:
Bumping this as building is broken in Ubuntu 22.04.

```
In file included from /home/dev/goestools/src/goesproc/map_drawer.h:5,
                 from /home/dev/goestools/src/goesproc/handler_goesn.cc:12:
/home/dev/goestools/src/goesproc/proj.h:16:10: fatal error: proj_api.h: No such file or directory
   16 | #include <proj_api.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:174: src/goesproc/CMakeFiles/goesproc.dir/handler_goesn.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:2398: src/goesproc/CMakeFiles/goesproc.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

pietern, 2024-02-10:
This was fixed in #148 and #167.

closed 2024-02-10 by pietern