pietern/goestools #131: Having troubles getting GOESnoncmip.conf to work

Mopar44084, 2022-02-18:
I followed "http://usradioguy.com/goes-satellite-imagery-reception/#enable-it/" tutorial to add noncmip.conf
it follows with no directory 

I created the GOESnoncmip.conf in /home/pi/goes/
I then ran "goesproc -c GOESnoncmip.conf -m packet –subscribe tcp://127.0.0.1:5004 –out /home/pi/goes/"
said no file or directory

Is there a way you can send me a step by step process if the site has something missed spelled?
Thanks!
Chris

Volas171, 2022-02-19:
you must run it on the same directory that GOESnoncmip.conf is located, else just use something like this 
`goesproc -c /home/pi/GOESnoncmip.conf -m packet –subscribe tcp://127.0.0.1:5004 –out /home/pi/goes/`
**NOTICE: YOU ARE RUNNING ONLY GOESnoncmip.conf YOU WONT GET THE OTHER IMAGERY UNLESS YOU ADD THE CONFIGURATION TO THE DEFAULT CONFIG** (simply add what is inside of GOESnoncmip.conf to  /usr/share/goestools/goesproc-goesr.conf then run 
`goesproc -c  /usr/share/goestools/goesproc-goesr.conf -m packet –subscribe tcp://127.0.0.1:5004 –out /home/pi/goes/`
should fix the problems you are having
Sam

Mopar44084, 2022-02-19:
Ok, I put the GOES noncmip.conf in the /home/pi/
  
Next I also copied all the files in the goesproc-goesr.conf
  

  
Then I got a error about the directory then -o argument
  
(Out) so I took command out completely out. And it seems to work. It created directory noncmip then a bunch other directories

creinemann, 2022-02-19:
The reason I have to keep the nonCMIP out of the goesproc-goesr.conf files is to have control over what is received. As I do not want NonCMIP imagery all the time.  My goesproc-goesr.conf file is running 24/7 and is started in rc.local, then if I want CMIP then I just run 
`goesproc -c GOESnoncmip.conf -m packet --subscribe tcp://127.0.0.1:5004 --out /home/pi/goes`
or EMWIN
`goesproc -c goesemwin.conf -m packet --subscribe tcp://127.0.0.1:5004 --out /home/pi/goes`
or HurricaneEnhancement
`goesproc -c goesNHC.conf -m packet --subscribe tcp://127.0.0.1:5004 --out /home/pi/goes`
or any of the other processes I have written handlers for.

The out command specifies where I want the files stored based on the handler of the particular process.  Example: no sense in running the hurricane enhancement when it's not hurricane/cyclone season.

Mopar44084, 2022-02-19:
Thanks that helps out a bunch!, I might create a new goesproc-goesr.conf when I want to receive everything. I will play around with it.
  

  
Do you have a guide what everything means?
  
Fd
  
Ch02-ch15
  
M1
  
M2   
  
Etc

creinemann, 2022-02-19:
Yes I have multiple pages of documentation at
 (https://usradioguy.com/goes-satellite-imagery-reception/)
(https://usradioguy.com/online-satellite-imagery-and-information/)
(https://usradioguy.com/goes-band-imagery/)

closed 2022-03-11 by Mopar44084

CheaterAdams, 2025-05-24:
With the transition from GOES-16 to GOES-19 I have updated goesproc-goesr.conf with the new conf file that includes goes19 and it is working fine using command 
   'goesproc -c /usr/share/goestools/goesproc-goesr.conf -m packet --subscribe tcp://192.168.1.207:5004 --out $GOESHOME'.
Emwin decode also works.
I copied my original (successfully used with goes16) GOESnoncmip.conf. My GOES19noncmip conf file is located in ~/goes conf.d and the command I use is "goesproc -c ~/goes/conf.d/GOES19noncmip.conf -m packet --subscribe tcp://192.168.1.207:5004 --out $GOESHOME". 
Output shows the packet counter line (Packet SCID=0 VCID=13 counter=xxxxxx) but no images are written.
Questions:
    1 Should the noncmip config file have handlers for goes19?
   2. Should the noncmip config used for goes16 work as is for goes19?.