pietern/goestools #132: How to receive Goes16 northern Rockies images

Mopar44084, 2022-02-18:
How do I receive images from northern Rockies?
In the Goes16 folder m1 & M2 I only see Texas and Tennessee.

Thanks,
Chris

rpdale, 2022-02-18:
Those are the mesoscale floater images which are focused on active weather. You would want the conus images.

Mopar44084, 2022-02-18:
Is that possible to receive with goestools on the Raspberry pi 4 8gb model?

creinemann, 2022-02-19:
Yes, 
Mesoscale imagery is determined by NOAA, NWS, etc. Meso Scale locations are centered on an active area, and they switch quite often. CONUS is not available as an HRIT image from either GOES, you would have to create your own custom crops to get a pseudo mesoscale image of just the rockies.

closed 2022-03-11 by Mopar44084