pietern/goestools #136: Update goestools to receive Himawari-9

pietern, 2022-03-16:
From #122:
> Himawari 9 transition from Himawari 8 WAS scheduled for April 2022, but has been pushed to December 2022, Again, some updates need to be made, such as src/goesproc/handler_himawari8.cc It is not known as of yet if the NOAA LRIT header file will remain '43' or be changed.

This is blocked on knowing what the LRIT product ID is going to be.

@creinemann Let's use a separate issue for Himawari-9 as it's not yet known what the work is going to be.

creinemann, 2022-11-30:
@pietern I sent you an email on the Himawari 9 update on 11/28/22  The HRIT Manager at NOAA had some questions about the new product ID for Himawari 9

Carl

SI-jvictors, 2022-12-13:
This is supposed to happen today, are we ready?