pietern/goestools #141: EMWIN handler not parsing time properly

RyokoAsakura, 2022-05-16:
It would appear that I am having the same problem in my EMWIN folder/date creation as was occurring for nws in issue #100. Is there any way to fix the emwin handler so this doesn't happen?

Thank you in advance.

[image]

ZL1LAC, 2022-10-29:
I don't seem to be having this issue, can you confirm you are still getting this problem @RyokoAsakura?

RyokoAsakura, 2022-10-31:
I can confirm that I am still having this issue. So much, in fact, that I still have a script cron'ing a file copy from the folder every 2 minutes.

ZL1LAC, 2022-11-15:
Have you pulled the latest updates?  I'm not having this issue so maybe you're still on a old build?

RyokoAsakura, 2022-11-15:
Just pulled, now running latest update (865e5c7) and the problem still exists.

ZL1LAC, 2022-11-15:
Can you share your goesproc conf file so I can compare to mine.

RyokoAsakura, 2022-11-15:
[goesproc-goesr.conf.txt](https://github.com/pietern/goestools/files/10016917/goesproc-goesr.conf.txt)

ZL1LAC, 2022-11-16:
Just compared uses to mine and the only thing I see that's diffrent is theres a "." in my path at the start.  Maybe try doing that as well?

```
[[handler]]
type = "emwin"
directory = "./emwin/{time:%Y-%m-%d}"
exclude_txt = true
```

RyokoAsakura, 2022-11-16:
No change. All that does is tell it to use the current directory.
[image]

RyokoAsakura, 2022-11-16:
I appreciate your effort. Thanks!

RyokoAsakura, 2023-09-21:
Personal solution until it gets fixed:

I have written 2 scripts that moves the 1969-12-31 in EMWIN to the proper place. The one script is for the current UTC day, and the second script is for whenever your local day is after UTC 00:00. I run them via cron job every 2 minutes during appropriate hours. The only thing you have to do is adjust the hours they get run based on when your local time is UTC 00:00. 

Here's my cron layout for UTC+5 (US Eastern Time)
*/2 0-19/1 * * * bash /home/pi/move1969.sh
*/2 20-23/1 * * * bash /home/pi/nextdaymove1969.sh

And attached are the bash scripts for both. Hope these can help. Don't forget to remove the .txt and also to chmod +x the scripts before you set the cron job for them.

--Ryoko

[move1969.sh.txt](https://github.com/pietern/goestools/files/12678904/move1969.sh.txt)
[nextdaymove1969.sh.txt](https://github.com/pietern/goestools/files/12678905/nextdaymove1969.sh.txt)