pietern/goestools #142: fatal error: proj_api.h: No such file or directory when building goestools under Ubuntu 22.04

cosmopotamian, 2022-05-17:
sudo make install

[  0%] Building CXX object src/util/CMakeFiles/util.dir/fs.cc.o
[  1%] Building CXX object src/util/CMakeFiles/util.dir/string.cc.o
[  1%] Building CXX object src/util/CMakeFiles/util.dir/time.cc.o
[  1%] Linking CXX static library libutil.a
[  1%] Built target util
[  2%] Building CXX object src/lib/CMakeFiles/dir.dir/dir.cc.o
[  2%] Linking CXX static library libdir.a
[  2%] Built target dir
[  2%] Building CXX object src/lib/CMakeFiles/zip.dir/zip.cc.o
[  3%] Linking CXX static library libzip.a
[  3%] Built target zip
[  4%] Building CXX object src/lib/CMakeFiles/timer.dir/timer.cc.o
[  4%] Linking CXX static library libtimer.a
[  4%] Built target timer
[  4%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/ep.c.o
[  4%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/global.c.o
[  5%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/pipe.c.o
[  5%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/poll.c.o
[  5%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/sock.c.o
[  7%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/sockbase.c.o
[  7%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/symbol.c.o
[  7%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/ctx.c.o
[  8%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/fsm.c.o
[  8%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/pool.c.o
[  9%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/timer.c.o
[  9%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/timerset.c.o
[  9%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/usock.c.o
[ 10%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/worker.c.o
[ 10%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/alloc.c.o
[ 10%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/atomic.c.o
[ 11%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/chunk.c.o
[ 11%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/chunkref.c.o
[ 11%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/clock.c.o
[ 13%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/closefd.c.o
[ 13%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/efd.c.o
[ 14%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/err.c.o
[ 14%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/hash.c.o
[ 14%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/list.c.o
[ 15%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/msg.c.o
[ 15%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/condvar.c.o
[ 15%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/mutex.c.o
[ 16%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/once.c.o
[ 16%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/queue.c.o
[ 16%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/random.c.o
[ 17%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/sem.c.o
[ 17%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/sleep.c.o
[ 19%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/strcasecmp.c.o
[ 19%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/strcasestr.c.o
[ 19%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/strncasecmp.c.o
[ 20%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/thread.c.o
[ 20%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/wire.c.o
[ 20%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/devices/device.c.o
[ 21%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/utils/dist.c.o
[ 21%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/utils/excl.c.o
[ 21%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/utils/fq.c.o
[ 22%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/utils/lb.c.o
[ 22%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/utils/priolist.c.o
[ 23%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/bus/bus.c.o
[ 23%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/bus/xbus.c.o
[ 23%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/pipeline/push.c.o
[ 25%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/pipeline/pull.c.o
[ 25%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/pipeline/xpull.c.o
[ 25%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/pipeline/xpush.c.o
[ 26%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/pair/pair.c.o
[ 26%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/pair/xpair.c.o
[ 26%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/pubsub/pub.c.o
[ 27%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/pubsub/sub.c.o
[ 27%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/pubsub/trie.c.o
[... 178 lines cut ...]
[ 86%] Built target goesrecv
[ 86%] Building CXX object src/goesrecv/CMakeFiles/benchmark.dir/benchmark.cc.o
[ 88%] Linking CXX executable benchmark
[ 88%] Built target benchmark
[ 89%] Building CXX object src/goeslrit/CMakeFiles/goeslrit.dir/goeslrit.cc.o
[ 89%] Building CXX object src/goeslrit/CMakeFiles/goeslrit.dir/options.cc.o
[ 89%] Linking CXX executable goeslrit
[ 89%] Built target goeslrit
[ 90%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/area.cc.o
[ 90%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/config.cc.o
[ 91%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/filename.cc.o
[ 91%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/file_writer.cc.o
[ 91%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/goesproc.cc.o
[ 92%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/gradient.cc.o
[ 92%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/handler_emwin.cc.o
[ 92%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/handler_goesn.cc.o
In file included from /home/engineering/Documents/goestools/src/goesproc/map_drawer.h:5,
                 from /home/engineering/Documents/goestools/src/goesproc/handler_goesn.cc:12:
/home/engineering/Documents/goestools/src/goesproc/proj.h:16:10: fatal error: proj_api.h: No such file or directory
   16 | #include <proj_api.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:174: src/goesproc/CMakeFiles/goesproc.dir/handler_goesn.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2398: src/goesproc/CMakeFiles/goesproc.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

x31415926, 2022-05-22:
I don't know this is the best way to fix it since I'd expect both version will need to coexist for a while. So I'd defer to the repository owner on whether this is ideal or not. On top of it, my testing was limited, but here's the patch diff I used in case someone else wants to try it out.
[proj8.patch.zip](https://github.com/pietern/goestools/files/8748982/proj8.patch.zip)

twcutter, 2022-05-23:
I'm assuming you are using debian? The current version of MIT's PROJ library is 9.0.0[(https://proj.org) which is not backward compatible.  The last version which had the proj_api.h was probably PROJ.4 of libproj-dev found in buster release.  (https://packages.debian.org/buster/amd64/libproj-dev/filelist)   You might want to get the older library and remove the latest?

twcutter, 2022-05-23:
Now that I'm paying more attention .. ubuntu.. it appears in the bionic (18.04) release to be present which was libproj-dev  bionic, not in Jammy (22.04).

x31415926, 2022-05-23:
I am not sure who this question was addressed to (me or the issue creator), but, yes, the issue has to do with proj library no longer supporting a backward compatibility facility which enabled goestool to continue building on Debian. This manifests in most recent Ubuntu releases today and, at some point, will for other distros. The proj deprecation was on schedule for a while and the patch above tries to deal with the somewhat minimal changes required to port it to the most recent version.

gvfd107, 2022-07-27:
Same issue here for me, i tried deleting the dependencies for proj_api.h and resulted in many more fatal errors for the "Make" step. will revert back to Bionic until we can figure this out.

pietern, 2024-02-10:
This was fixed in #148 and #167. Thanks for reporting.

closed 2024-02-10 by pietern