pietern/goestools #143: Not receiving Himwari-8 or Meteosat

Mopar44084, 2022-06-13:
I am using my satellite on GOES 16. In the /home/pi/ directory I do not see Himwari-8 or Meteosat folder. 
I did enable EMWIN and I do see it in that folder. I have updated goestools.

zeke800, 2022-08-20:
I think Himawari-8 is only broadcast on GOES-17. Check if it's in your visibility. The same thing happened to me and I was very disappointed until I found that out.

Mopar44084, 2022-08-20:
I'm receiving Himawari-8 through GOES-16 EMWIN about every hour.

JVital2013, 2022-08-20:
> I'm receiving Himawari-8 through GOES-16 EMWIN about every hour.

This is correct, but high resolution imagery is not available in its own folder. It's only available in low resolution EMWIN folder on GOES-16.

As far as I know, imagery that comes in on the EMWIN virtual channels cannot be split out into their own folders without secondary scripts