pietern/goestools #150: Can not build on Ubuntu 22

dive76, 2022-10-01:
I get the following errors:

[ 45%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/poller.c.o
In file included from /home/lars/goestools/vendor/nanomsg/src/aio/poller.c:27:
/home/lars/goestools/vendor/nanomsg/src/aio/poller_epoll.inc: In function ‘nn_poller_add’:
/home/lars/goestools/vendor/nanomsg/src/aio/poller_epoll.inc:64:9: warning: unused variable ‘rc’ [-Wunused-variable]
   64 |     int rc;
      |         ^~
[ 46%] Linking C static library ../libnanomsg.a

....
[ 92%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/handler_goesn.cc.o
In file included from /home/lars/goestools/src/goesproc/map_drawer.h:5,
                 from /home/lars/goestools/src/goesproc/handler_goesn.cc:12:
/home/lars/goestools/src/goesproc/proj.h:16:10: fatal error: proj_api.h: No such file or directory
   16 | #include <proj_api.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:174: src/goesproc/CMakeFiles/goesproc.dir/handler_goesn.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2398: src/goesproc/CMakeFiles/goesproc.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

CMAKE looks ok:

lars@SatPC:~/goestools/build$ cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local
CMake Deprecation Warning at vendor/libcorrect/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Inline keyword found - inline
-- Restrict keyword found - restrict
CMake Deprecation Warning at vendor/nanomsg/CMakeLists.txt:27 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Detected nanomsg ABI v5.1.0
-- OS System is Linux
-- OS Version is 5.15.0-48-generic
-- Checking for module 'opencv'
--   No package 'opencv' found
-- Checking for module 'opencv4'
--   Found opencv4, version 4.5.4
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lars/goestools/build

TheBrinkOfTomorrow, 2022-12-15:
There's a fix here, .in this PR...  https://github.com/pietern/goestools/pull/148

It worked for me on Ubuntu 22.04.1 LTS...

pietern, 2024-02-10:
This fix has been merged. Thanks for reporting.

closed 2024-02-10 by pietern